espressif/esp-rainmaker #90: Esp Rainmaker as a component of the Esp32 S2 HMI

darrinbolden, 2021-07-28:
Hi Guys,
Has anyone had any success in using Rainmaker as a component of the Esp32 S2 HMI Dev Kit 1?

shahpiyushv, 2021-07-29:
@darrinbolden , can you be a bit more specific? ESP32-S2 HMI Dev Kit seems to be a development board whereas ESP RainMaker is a software project that works with ESP IDF. If you already have your own ESP IDF based project, you may add this line in your CMakeLists.txt file:

```
set(EXTRA_COMPONENT_DIRS ${RMAKER_PATH}/components ${RMAKER_PATH}/examples/common ${RMAKER_PATH}/components/esp-insights/components)
```

You can define RMAKER_PATH as an environment variable using `export RMAKER_PATH=/path/to/esp-rainmaker` and then add `set(RMAKER_PATH $ENV{RMAKER_PATH})` in your  CMakeLists.txt .

You can check any of the esp-rainmaker example's [CMakeLists.txt](https://github.com/espressif/esp-rainmaker/blob/master/examples/fan/CMakeLists.txt) for reference.

darrinbolden, 2021-08-07:
@shahpiyushv your observation is correct. My objective is to use Esp Rainmaker to control the Esp32 S2 HMI Dev kit 1 smart panel instead of the Q cloud /WeChat example given.  It appears that:  https://github.com/espressif/esp-dev-kits/blob/master/esp32-s2-hmi-devkit-1/examples/smart-panel/main/ui/ui_app.c Line 233 needs to be modified.  Esp Rainmaker will operate in SoftAP mode. I will insert the Esp Rmkr ENDPOINT and work on the build implementing the earlier suggestion made (CMakeLists.txt). I will post the log output on this thread using Windows 10 in addition to the latest ESP IDF. v4.3 and Esp Rainmaker v2.0 as a component of the  Esp32 S2 HMI Dev kit 1 (the latest stable release).
 asprintf(&qr_text,
        "https://iot.cloud.tencent.com/iotexplorer/device?page=softap&productId=%s&ver=%s&name=%s",
        CONFIG_QCLOUD_PRODUCT_ID, "v1", CONFIG_LIGHT_PROVISIONING_SOFTAPCONFIG_SSID);