espressif/esp-rainmaker #92: Enabling insights crashes the "switch" example

ajlennon, 2021-08-04:
I followed the instructions to enable Rainmaker insights

https://rainmaker.espressif.com/docs/esp-insights.html

I updated the firmware via OTA (I also reflashed and re-provisoned) but I get this

```
I (1023) wifi_init: WiFi RX IRAM OP enabled
I (1043) esp_rmaker_work_queue: Work Queue created.
I (1063) esp_mqtt_glue: Initialising MQTT
I (1063) app_main: RainMaker Initialised.
I (1073) esp_rmaker_node: Node ID ----- pcsC3F5K7XrGfmERBXCBvV
I (1073) esp_rmaker_ota: OTA state = 2
I (1073) esp_rmaker_ota_using_params: OTA enabled with Params
E (1083) esp_rmaker_ota: OTA already initialised
I (1083) esp_rmaker_time_service: Time service enabled
I (1093) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (1103) esp_insights: ESP Insights enabled for Node ID ----- pcsC3F5K7XrGfmERBXCBvV
I (1113) app_insights: App Insights Enabled.
I (1113) esp_rmaker_core: Starting RainMaker Work Queue task
I (1123) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (1123) event: handler already registered, overwriting
E (1133) esp_insights: Failed to initialise RTC store.
I (1143) app_insights: App Insights Enabled.
```

```
I (3983) esp_rmaker_time: SNTP Synchronised.
I (3983) esp_rmaker_time: The current time is: Wed Aug  4 22:38:42 2021 +0800[CST], DST: No.
I (6783) esp_mqtt_glue: MQTT Connected
I (6783) esp_rmaker_node_config: Reporting Node Configuration of length 1199 bytes.
I (6793) esp_rmaker_param: Reporting params (init): {"Switch":{"Name":"Switch","Power":true},"OTA":{"Status":"","Info":"","URL":""},"Time":{"TZ":"","TZ-POSIX":""},"Schedule":{"Schedules":[]}}
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c46c  PS      : 0x00060e30  A0      : 0x800e1eea  A1      : 0x3ffdb210  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001000  A5      : 0x00000000  
A6      : 0x400e1f98  A7      : 0x00000100  A8      : 0x8016ef40  A9      : 0x3ffdb1f0  
0x400e1f98: insights_periodic_handler at /data_drive/git/simpson/esp-rainmaker/components/esp-insights/components/esp_insights/src/esp_insights.c:330

A10     : 0x3ffb9500  A11     : 0xffffffff  A12     : 0x00000006  A13     : 0x3ffcedd4  
A14     : 0x40081824  A15     : 0x3ffc4b8c  SAR     : 0x00000019  EXCCAUSE: 0x0000001d  
0x40081824: task_ms_to_tick_wrapper at /data_drive/git/simpson/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:398

EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x000000ff  

Backtrace:0x4000c469:0x3ffdb210 0x400e1ee7:0x3ffdb220 0x400e1fa7:0x3ffdb250 0x400e409c:0x3ffdb2c0 0x400e40d1:0x3ffdb2f0 0x40094ada:0x3ffdb310
0x400e1ee7: send_insights_data at /data_drive/git/simpson/esp-rainmaker/components/esp-insights/components/esp_insights/src/esp_insights.c:287

0x400e1fa7: insights_periodic_handler at /data_drive/git/simpson/esp-rainmaker/components/esp-insights/components/esp_insights/src/esp_insights.c:341

0x400e409c: esp_rmaker_handle_work_queue at /data_drive/git/simpson/esp-rainmaker/components/esp-insights/components/rmaker_common/src/work_queue.c:49

0x400e40d1: esp_rmaker_work_queue_task at /data_drive/git/simpson/esp-rainmaker/components/esp-insights/components/rmaker_common/src/work_queue.c:58

0x40094ada: vPortTaskWrapper at /data_drive/git/simpson/esp/esp-idf/components/freertos/port/xtensa/port.c:168
```

If then continuously resets itself. 

Any thoughts?

shubhamdp, 2021-08-04:
@ajlennon Looks like `app_insights_enable()` is being called twice. Have you added a call to `app_insights_enable()` in `examples/switch/main/app_main.c`? If yes then that is not required. As mentioned [here](https://rainmaker.espressif.com/docs/esp-insights.html#enable-the-esp-insights), it is already included in all ESP RainMaker examples. So, you will just have to enable CONFIG_ESP_INSIGHTS_ENABLED.

ajlennon, 2021-08-04:
Silly me. You are absolutely right! I fixed that and it boots up. 

It sends this

```
I (6693) esp_insights: Sending data of length 121 to the MQTT Insights topic:

0x02 0x76 0x00 0xa1 0x64 0x64 0x69 0x61 0x67 0xbf 0x63 0x76 0x65 0x72 0x63 0x31 
0x2e 0x30 0x62 0x74 0x73 0x1a 0x00 0x5d 0x21 0x57 0x66 0x73 0x68 0x61 0x32 0x35 
0x36 0x70 0x30 0x39 0x65 0x30 0x33 0x35 0x62 0x30 0x32 0x65 0x61 0x30 0x64 0x64 
0x64 0x36 0x64 0x64 0x61 0x74 0x61 0xbf 0x64 0x62 0x6f 0x6f 0x74 0xbf 0x62 0x74 
0x73 0x1a 0x00 0x03 0x1c 0x10 0x64 0x63 0x68 0x69 0x70 0x01 0x68 0x63 0x68 0x69 
0x70 0x5f 0x72 0x65 0x76 0x01 0x66 0x72 0x65 0x61 0x73 0x6f 0x6e 0x01 0x64 0x70 
0x72 0x6f 0x6a 0x66 0x73 0x77 0x69 0x74 0x63 0x68 0x67 0x61 0x70 0x70 0x5f 0x76 
0x65 0x72 0x63 0x31 0x2e 0x32 0xff 0xff 0xff 

```
But I am not seeing anything on the dashboard? I just get

"Please ensure that the diagnostics support is enabled in your firmware to turn on insights."

shubhamdp, 2021-08-05:
@ajlennon Can you please share the screenshot of the webpage. Also, on which pages do you see this message? There are three(Node Reports, Metrics, and Variables).

ajlennon, 2021-08-05:
Sure so this is the main page which looks ok

[image]

And this is where I expected more info

[image]

shubhamdp, 2021-08-05:
I assume you are using the `examples/switch` as is, it does not have all the functionality enabled. Also, I see there is some data on the Node reports page that means you are all set. For [metrics](https://github.com/espressif/esp-insights#metrics) and [variables](https://github.com/espressif/esp-insights#variables) you'll have to enable few config options.

Please read more [here](https://github.com/espressif/esp-insights). Also, please have a look at [ESP Insights features](https://github.com/espressif/esp-insights#features) set.

For some quick tests, you can add error logs(`ESP_LOGE`) or events([ESP_DIAG_EVENT](https://github.com/espressif/esp-insights/blob/main/components/esp_diagnostics/include/esp_diagnostics.h#L202-L206)) , these logs should be visible on the Node reports page.
Enable `CONFIG_DIAG_ENABLE_METRICS=y` and `CONFIG_DIAG_ENABLE_HEAP_METRICS=y` config options to view heap metrics on the console. On metrics page, for the first time you'll have to select the filter by clicking plus button(one on the right side).

shahpiyushv, 2021-08-06:
@ajlennon , it would be hard to suggest without even knowing what the use case it. This specific issue was opened for "Insights" which is a diagnostics solutions whereas RainMaker is more of an end to end IoT system. The EduKit example which leverages RainMaker is just one example based on RainMaker, built for a specific development board.

So most probably, The "switch" example in ESP RainMaker would still be the right place to get started, which is what I see that you have already done. The next step depends on what functionality you are looking for. If you have any commercial enquiries related to RainMaker, you can get in touch over email on sales@espressif.com.

ajlennon, 2021-08-06:
Hi @shahpiyushv,

>@ajlennon , it would be hard to suggest without even knowing what the use case it. This specific issue was opened for "Insights" which is a diagnostics solutions whereas RainMaker is more of an end to end IoT system.

Specifically wrt Insights, I started here, with no reference to AWS IoT

https://rainmaker.espressif.com/docs/get-started.html

Then I went through the rest of the pages there until I got to this

ajlennon, 2021-08-06:
Aha - "minimal_diagnostics" ? Thanks! Will try...

shubhamdp, 2021-08-06:
@ajlennon Please take a look [here](https://github.com/espressif/esp-insights/tree/main/examples), there are two examples which demonstrates how insights can be used.

shahpiyushv, 2021-08-06:
@ajlennon , As @shubhamdp has mentioned, if you are interested purely in "ESP Insights", which is a diagnostics solution that would typically work with your existing or new IoT application, you can check the examples [here](https://github.com/espressif/esp-insights/tree/main/examples). However, if you want to leverage AWS IoT to develop your own end to end IoT system itself, you should start with any of the RainMaker examples and then (optionally) enable Insights as required for diagnostics.

ajlennon, 2021-08-06:
That's great - thanks!