espressif/esp-rainmaker #94: Scheduling not supported in homekit_example

adukale, 2021-08-06:
Hi

Is there any specific reason scheduling is not supported in homekit_example.
Or do I need to enable some flag to enable scheduling?

Thanks in advance.

shahpiyushv, 2021-08-06:
Scheduling is an optional feature and not relevant not directly relevant in the context of a demo app that's primarily to show how RainMaker + HomeKit works together. That's the only reason why it's not there. You can read the docs [here] (https://rainmaker.espressif.com/docs/scheduling.html#usage) or copy relevant code from any of the examples (including timezone related code) and enable scheduling.

adukale, 2021-08-08:
@shahpiyushv Noted. Will check.
Thanks for the information.

closed 2021-08-08 by adukale