espressif/esp-rainmaker #95: set-target esp32s2

ajsb85, 2021-10-02:
Dearest.

The examples should have the setup for the ESP32-S2...

```console
PowerShell 7.1.4
Copyright (c) Microsoft Corporation.

https://aka.ms/powershell
Type 'help' to get help.

PS C:\dev\GitHub\esp-rainmaker\examples\switch> idf.py set-target esp32s2
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\dev\github\esp-rainmaker\examples\switch\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s2, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\dev\github\esp-rainmaker\examples\switch\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=1 c:\dev\github\esp-rainmaker\examples\switch"...
-- Found Git: C:/Users/ajsb85/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
CMake Error at C:/Users/ajsb85/.esp-idf/tools/cmake/targets.cmake:19 (message):
  IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.
  To change the target, clear the build directory and sdkconfig file, and
  build the project again
Call Stack (most recent call first):
  C:/Users/ajsb85/.esp-idf/tools/cmake/project.cmake:38 (__target_init)
  CMakeLists.txt:15 (include)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
PS C:\dev\GitHub\esp-rainmaker\examples\switch> 
```

shahpiyushv, 2021-10-04:
As indicated in the error message *To change the target, clear the build directory and sdkconfig file, and
  build the project again*.
Can you just clear these files/folders?

```
$ rm -rf build/ sdkconfig sdkconfig.old
```

shahpiyushv, 2021-11-24:
Closing this issue since it is stale. Please update if the issue still persists.

closed 2021-11-24 by shahpiyushv