espressif/esp-rainmaker #102: ESP Insights not registered when new firmware with ESP Insights is flased on existing ESP rainmaker configured device

adukale, 2021-12-12:
The issues is, I had an esp rainmaker devices configured and installed for testing. This was configured and installed prior to inclusion of esp insights into rainmaker.
Now, I pushed an OTA to the device with `app_insights_enable();`. After OTA sucess the node is not showing itself in Rainmaker Dashboard. And in this process I changed OTA confin to topics so I can access the device now via CLI as well as dashboard.
Did I miss something or this is what the behaviour is, like esp insights only gets registered while initial configuration of rainmaker firmware.

shubhamdp, 2021-12-13:
Hi @adukale, can you please provide the following details
- Device (ESP32, S2, C3,..)
- If this is an ESP32-S2 was this self-claimed or claimed using rainmaker cli? ([ref for claiming](https://rainmaker.espressif.com/docs/claiming.html))
- Commit head for rainmaker

adukale, 2021-12-13:
@shubhamdp Here are the details:

- Device: ESP32
- Claiming used: Assisted claiming
- Commit Head: b516bf1

shubhamdp, 2021-12-13:
Thanks @adukale I'll check and get back.

shahpiyushv, 2021-12-13:
@adukale , do you mean that you do not see the node in the list of nodes in the dashboard at all? If yes, that is unrelated to ESP Insights as such, but a different issue altogether.

Even if Assisted Claiming is enabled, note that the claiming process repeats only if the ESP32's flash memory was completely erased. Otherwise it is very much possible that a node has different end user and admin user. You can check out the docs [here](https://rainmaker.espressif.com/docs/user-roles.html) for more information on this. If you are unable to find the node in any of your accounts, please share the node id with us over email on esp-rainmaker-admin@espressif.com.

Meanwhile, for ESP Insights, please also ensure that you have [CONFIG_ESP_INSIGHTS_ENABLED](https://rainmaker.espressif.com/docs/esp-insights.html#enable-the-esp-insights) enabled.

adukale, 2021-12-14:
@shahpiyushv Have sent email on esp-rainmaker-admin@espressif.com.
The same firmware was installed on another board, the insights were available for that.

My question is, Even if esp-insights is not enabled in idf.py menuconfig & in code then also the node will be visible in admin dashboard (dashboard.rainmaker.espressif.com) or not ?
If yes, then as you said this will be different issue altogether. Because this node is online from around August 5, 2021.

Also, I have a question, Let me know if I should open a new issue for this:
Can OTA_USING_PARAMS & OTA_USING_TOPICS enabled in same firmware ?

Thanks.

shahpiyushv, 2022-01-27:
As discussed on email, the issue was about the admin id being different for the nodes. 

Meanwhile, OTA_USING_PARAMS & OTA_USING_TOPICS cannot be enabled simultaneously.

closed 2022-01-27 by shahpiyushv