espressif/esp-rainmaker #104: ESP32 disconnects from Wifi after device is added in rainmaker app

sduddell, 2022-01-08:
Hi,

I'm trying to add a device in rainmaker app and edit one of its parameters.
But, after wifi provisioning and device gets added in the app, it gets disconnected from wifi.

Log messages look as shown in image:
[image]

My code looks something like below (similar to switch device example):

```
    /* Initialize the ESP RainMaker Agent.
     * */
    esp_rmaker_device_t *thing_device;

    esp_rmaker_config_t rainmaker_cfg = {
        .enable_time_sync = false,
    };
    esp_rmaker_node_t *node = esp_rmaker_node_init(&rainmaker_cfg, "ESP RainMaker Device", "Camera");
    if (!node) {
        ESP_LOGE(TAG, "Could not initialise node. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }

    thing_device = esp_rmaker_device_create("Thing", "esp.device.thing", NULL);

    esp_rmaker_device_add_cb(thing_device, write_cb, NULL);

    esp_rmaker_device_add_param(thing_device, esp_rmaker_name_param_create("name", "Thing"));

    //esp_rmaker_param_t *camera_param = esp_rmaker_param_create("param", NULL,
      //      esp_rmaker_str(""), PROP_FLAG_READ | PROP_FLAG_WRITE | PROP_FLAG_PERSIST);

    //esp_rmaker_device_add_param(thing_device,camera_param);

    /* Add this switch device to the node */
    esp_rmaker_node_add_device(node, thing_device);

    //esp_init();

    /* Start the ESP RainMaker Agent */
    esp_rmaker_start();

    result = app_wifi_start();
    if (result != ESP_OK) {
        ESP_LOGE(TAG, "Could not start Wifi. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }
```
Any help/suggestion will be appreciated

shahpiyushv, 2022-01-10:
@sduddell , do you see this with all Wi-Fi routers? Where are the Hello world prints coming from?

sduddell, 2022-01-11:
hi Piyush, thanks for your reply, I tried with two networks . One with my mobile hotspot and another with router wifi connection. I see this behavior with router network. Is there any minimum wifi RSSI for this to not happen?

closed 2022-01-13 by sduddell