espressif/esp-rainmaker #109: Enhancements to `led_light`

gtoombs-microsaas, 2022-01-24:
I have in hand a set of enhancements to the `led_light` example, including:

- Very simplistic exponential curvature brightness compensation for the WS2812
- Reuse of `esp_rmaker_param_val_t` instead of pulling them apart and putting them back together
- More consistent error handling and logging
- Factoring-out of duplicated code in `app_light_init`
- Replacement of string-comparison parameter checks with faster referential parameters checks
- Replacement of unrecognised-parameter ignore with proper logging

Is it worth submitting a PR?

shahpiyushv, 2022-01-25:
@gtoombs-microsaas , this indeed sounds useful. I believe this may not be applicable just for led_light, but other examples as well. It would help if you file separate PRs for individual features so that they are easier to review and merge.

reinderien, 2022-01-26:
First PR: https://github.com/espressif/esp-rainmaker/pull/110