espressif/esp-rainmaker #111: Mobile phone compatibility for RainMaker Android App using ESP32S2

eos1d3, 2022-02-12:
I tried the LED Light example using IDF Release 4.3. ESP32 boards work without issue. But ESP32S2 just cannot be added using Poco X3 NFC mobile phone. 

Another ESP-C3 with factory flashed LED Light example works with this mobile phone.

After scanning the QR Code, a small window showing:
```
Connect to device

ESP Rainmaker app wants to use a temporary Wi-Fi network to connect to your device
PROV-xxxxxx
```

After clicking Connect, a single line of message "Connection successful" is shown, but it jumps to another window showing:

Device connection failed.
Do you want to connect device manually

After clicking Yes, the app open a full screen for asking Connect to Device PROV_xxxxxx. Clicking Connect and get it connected, but RainMaker is still showing connecting and finally change the button to Connect again.

From my mobile, I find it really connected to PROV-xxxxxx. But I just can't add it even I tried other ESP32S2 boards, including SOLO and Saola.
 
The strange thing is another LG mobile phone works with these ESP32 devices. Rainmaker app just does not show the above windows, but showing Select Wi-Fi Network with a message:
`To complete setup of your device, PROV-xxxxxx, please provide your Home Network credentials.
`

The RainMaker app version is 2.4.1 - 21c2060 is used for Poco X3 NFC from PlayStore. APK direct link, version 2.0.2 97f5372 from Espressif is also used with exactly the same problem.

shahpiyushv, 2022-02-15:
@eos1d3 the most important difference between esp32c3 and esp32s2 that esp32c3 uses BLE provisioning whereas esp32s2 uses SoftAP. We indeed have seen that some Android variants do not work correctly with SoftAP, especially because this interface does not have Internet. Can you check your phone's Wi-Fi settings once to see if there is some "intelligent" switching enabled to ensure that the phone is always connected to the Internet?

eos1d3, 2022-02-17:
@shahpiyushv I have tried many times enabling "WiFI Assistant -Stay connected". This says to use mobile data when Wi-Fi network cannot connect to the internet. But no matter how many times I tried, the problem is the same.

But from WiFi setting, I can see my phone WiFi is connected to the soft SoftAP.