espressif/esp-rainmaker #112: ESP32-S3 Error 261. Failed to get PoP from NVS, Please perform Claiming

eos1d3, 2022-02-12:
ESP-IDF version: Release 4.4
ESP device: ESP32-S3
LED_Light example

There are errors after Assisted Claiming:

```
I (742) esp_claim: Initialising Assisted Claiming. This may take time.
W (752) esp_claim: Generating the private key. This may take time.
I (18292) esp_rmaker_node: Node ID ----- 7CDFA1E121C4
I (18302) esp_rmaker_ota: OTA state = 2
I (18302) esp_rmaker_ota_using_params: OTA enabled with Params
I (18302) esp_rmaker_time_service: Time service enabled
I (18302) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (18322) app_insights: Enable CONFIG_ESP_INSIGHTS_ENABLED to get Insights.
I (18322) esp_rmaker_core: Starting RainMaker Work Queue task
I (18332) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (18332) BT_INIT: esp_bt_mem_release not implemented, return OK
I (18332) esp_claim: Waiting for assisted claim to finish.
I (18342) wifi_prov_scheme_ble: BT memory released
I (18352) app_wifi: Starting provisioning
E (18362) app_wifi: Error: 261. Failed to get PoP from NVS, Please perform Claiming.
E (18362) app_main: Could not start Wifi. Aborting!!!
```

BTW, there are some warnings during compilation:

```
warning: 'portENTER_CRITICAL_NESTED' is deprecated 
warning: 'portEXIT_CRITICAL_NESTED' is deprecated
warning: 'vPortCPUInitializeMutex' is deprecated 
warning: passing argument 2 of 'esp_rmaker_post_event' discards 'const' qualifier from pointer target type 
```

shahpiyushv, 2022-02-14:
@eos1d3 , I just tried this with esp idf v4.4 release and esp-rainmaker master branch on an esp32s3. Erased the flashed and then used `idf.py flash monitor`. I get this output

```
I (697) esp_claim: Initialising Assisted Claiming. This may take time.
I (1007) esp_claim: Private key already exists. No need to re-initialise it.
I (1057) esp_rmaker_node: Node ID ----- 84F7038005F4
I (1057) esp_rmaker_ota: OTA state = 2
I (1057) esp_rmaker_ota_using_params: OTA enabled with Params
I (1057) esp_rmaker_time_service: Time service enabled
I (1067) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (1077) app_insights: Enable CONFIG_ESP_INSIGHTS_ENABLED to get Insights.
I (1077) esp_rmaker_core: Starting RainMaker Work Queue task
I (1087) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (1087) BT_INIT: esp_bt_mem_release not implemented, return OK
I (1097) esp_claim: Waiting for assisted claim to finish.
I (1107) wifi_prov_scheme_ble: BT memory released
I (1107) app_wifi: Starting provisioning
I (1117) phy_init: phy_version 302,3ddfab3,Dec  3 2021,14:39:12
I (1167) wifi:mode : sta (84:f7:03:80:05:f4)
I (1177) wifi:enable tsf
W (1177) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (1177) BT_INIT: BT controller compile version [6881869]
I (1177) BT_INIT: Bluetooth MAC: 84:f7:03:80:05:f6

I (1187) protocomm_nimble: BLE Host Task Started
I (1197) NimBLE: GAP procedure initiated: stop advertising.

I (1197) NimBLE: GAP procedure initiated: advertise; 
I (1207) NimBLE: disc_mode=2
I (1207) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (1217) NimBLE: 

I (1227) wifi_prov_mgr: Provisioning started with service name : PROV_77d8bb 
I (1227) app_wifi: Provisioning started
I (1237) esp_rmaker_local: Event 1
I (1237) app_wifi: Scan this QR code from the ESP RainMaker phone app for Provisioning.
                                      
  █▀▀▀▀▀█ ▄▄█   ▄█▄▄█▀▄▄▀█▄ █▀▀▀▀▀█   
  █ ███ █ ▀▄  ▄  ██▀█▄  ▀▄▄ █ ███ █   
  █ ▀▀▀ █ █▀█ █▀███ █▄ █▀▀█ █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ █ █▄▀ ▀ ▀▄▀ █▄█ █ ▀▀▀▀▀▀▀   
  █▀ ▀ ▄▀▀▄▄ ██▀▄▀█▄  ▄█ ▀ ▄█▀▀▄▀▀▄   
  ▀██▀▄▀▀█▀▄▄██▄█▄ ▀██▀█ ▄▄ ▄█    █   
   ▄▀ █ ▀█▄▄▀█ ▀▀  ▀█▄ █▄█▄▀█ ▄▄ ▀█   
  █▄██▀ ▀ ▄▀▀█ █   █▀  █▀▀█▀  ▄█ █▄   
   █▄▀▄█▀ ▄▄▄ █▄▄ █▀█▀██  ▄█▀ ▀▀▄ ▀   
  ▀▄█▄██▀ █  ▀█▀▀ ▀█▄▄▀██ ▄ █▄▄▄ ▀█   
   ▄▄█▀█▀▀▄ █▄ ▀▀▀█▄▀▀▀▀▀▄█ ▄▄ █ ▄▄   
  ▀██▀▄▄▀ ▀▄▀  █▄▄▀▀▀▀█▄▀▄▀█  ▄▄ ▀▀   
  ▀  ▀ ▀▀ ██ ▄█▀  ▀▀▀▄▀██ █▀▀▀█▀ ▄▄   
  █▀▀▀▀▀█ ▀█▄ █▄█ ▀█ ▄ █▀▀█ ▀ ██ █    
  █ ███ █ ▄█▄▀▀▄█ █▀▀▀ ██ ▀████▀ ▄█   
  █ ▀▀▀ █ ▄█ ▄▄▀▀▄▀█▀▄▄▄█ ▀  ▄▀█▀▀▀   
  ▀▀▀▀▀▀▀ ▀    ▀▀    ▀ ▀  ▀ ▀ ▀▀ ▀    
                                      

I (1437) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_77d8bb","pop":"4eaddc87","transport":"ble"}
I (1457) app_wifi: Provisioning Started. Name : PROV_77d8bb, POP : 4eaddc87
```

So, it is hard to tell what the problem  could be. Can you double check that your flash was erased before you flashed the firmware. Can you double check that your partition table is right? It is printed on boot-up

```
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (72) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (79) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (94) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (101) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (109) boot: End of partition table
```

eos1d3, 2022-02-14:
The board I am using is ESP32-S3-DevKitC-1, ordered from Espressif China directly.

This is my partition table:
```
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (74) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (81) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (89) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (96) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (104) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (111) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (119) boot: End of partition table

```

I do erase-flash not just once, but the problem is the same. Just tried again with new setup and still got the same problem.

```
I (692) wifi_init: rx ba win: 6
I (702) wifi_init: tcpip mbox: 32
I (702) wifi_init: udp mbox: 6
I (702) wifi_init: tcp mbox: 6
I (712) wifi_init: tcp tx win: 5744
I (712) wifi_init: tcp rx win: 5744
I (722) wifi_init: tcp mss: 1440
I (722) wifi_init: WiFi IRAM OP enabled
I (722) wifi_init: WiFi RX IRAM OP enabled
I (742) esp_rmaker_work_queue: Work Queue created.
I (742) esp_claim: Initialising Assisted Claiming. This may take time.
W (752) esp_claim: Generating the private key. This may take time.
I (4772) esp_rmaker_node: Node ID ----- 7CDFA1E121C4
I (4782) esp_rmaker_ota: OTA state = 2
I (4782) esp_rmaker_ota_using_params: OTA enabled with Params
I (4782) esp_rmaker_time_service: Time service enabled
I (4782) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (4792) app_insights: Enable CONFIG_ESP_INSIGHTS_ENABLED to get Insights.
I (4802) esp_rmaker_core: Starting RainMaker Work Queue task
I (4802) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (4802) BT_INIT: esp_bt_mem_release not implemented, return OK
I (4812) esp_claim: Waiting for assisted claim to finish.
I (4822) wifi_prov_scheme_ble: BT memory released
I (4832) app_wifi: Starting provisioning
E (4842) app_wifi: Error: 261. Failed to get PoP from NVS, Please perform Claiming.
E (4842) app_main: Could not start Wifi. Aborting!!!
```

I found other has the same problem from ESP32.com:
https://www.esp32.com/viewtopic.php?f=25&t=25649

eos1d3, 2022-02-14:
My board shall be  ESP32-S3-DevKitC-1-N32R8V (32MB Flash, 8MB PSRAM)
 
Can the large flash size cause the problem? See the two warnings below:

```
I (491) heap_init: Initializing. RAM available for dynamic allocation:
I (498) heap_init: At 3FCA1B38 len 0003E4C8 (249 KiB): D/IRAM
I (505) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (511) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (517) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (525) spi_flash: detected chip: mxic
W (528) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (539) spi_flash: flash io: dio
W (543) spi_flash: Detected size(32768k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (557) sleep: Configure to isolate all GPIO pins in sleep state
I (563) sleep: Enable automatic switching of GPIO sleep configuration``
```

shahpiyushv, 2022-02-15:
@eos1d3 The flash related warning just says "access beyond 16 MB is not supported for this flash model" which I suppose should be fine since the partition table is for 4MB. Can you enable debug logs and see what additional information it gives? You can do so using: idf.py menuconfig -> Component config -> Log output -> Default log verbosity -> Debug

shahpiyushv, 2022-03-16:
@eos1d3 , can you pull the latest esp-idf and esp-rainmaker and retry? [This commit](https://github.com/espressif/esp-idf/commit/12e050b38c373cfaf67ade4b65f0afa8d50b92d8) in esp-idf should help resolve this issue.

eos1d3, 2022-03-17:
@shahpiyushv I just made clean installation of IDF master and esp-rainmaker. Tested with `led_light` example and erase flash before flashing new binary. It gives the same problem. But I am seeing other warnings:

```
W (460) flash HPM: Enter HPM by reconfiguring dummy has not been fully tested
E (460) flash HPM: Flash high performance mode hasn't been supported

Warning: checksum mismatch between flashed and built applications. Checksum of built application is 01496fd2958f649a7881bca3d5fbca78d4e441c08c3b99d397610b2bce5ec171

```

And the same HPM error happens in `blink` example too.  Does my board (ESP32-S3-DevKitC-1-N32R8V, 32MB Flash, 8MB PSRAM) has new EEPROM which is not yet supported?

The full log here:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0110,len:0x18d0
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x319c
entry 0x403b6240
I (33) boot: ESP-IDF v5.0-dev-2046-g5963de1caf 2nd stage bootloader
I (33) boot: compile time 11:40:55
I (33) boot: chip revision: 0
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : SLOW READ
I (47) boot.esp32s3: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (75) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (82) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (90) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (97) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (105) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (112) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (120) boot: End of partition table
I (124) boot: No factory image, trying OTA 0
I (129) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=3ed6ch (257388) map
I (198) esp_image: segment 1: paddr=0005ed94 vaddr=3fc979d0 size=01284h (  4740) load
I (200) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=dce00h (904704) map
I (417) esp_image: segment 3: paddr=0013ce28 vaddr=3fc98c54 size=03b90h ( 15248) load
I (421) esp_image: segment 4: paddr=001409c0 vaddr=40374000 size=139c4h ( 80324) load
I (444) esp_image: segment 5: paddr=0015438c vaddr=50000000 size=00010h (    16) load
I (453) boot: Loaded app from partition at offset 0x20000
I (481) boot: Set actual ota_seq=1 in otadata[0]
I (481) boot: Disabling RNG early entropy source...
W (492) flash HPM: Enter HPM by reconfiguring dummy has not been fully tested
E (492) flash HPM: Flash high performance mode hasn't been supported
I (495) cpu_start: Pro cpu up.
I (499) cpu_start: Starting app cpu, entry point is 0x40375428
0x40375428: call_start_cpu1 at /Users/andy/esp/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (521) cpu_start: Pro cpu start user code
I (521) cpu_start: cpu freq: 160000000 Hz
I (521) cpu_start: Application information:
I (524) cpu_start: Project name:     led_light
I (529) cpu_start: App version:      1.0
I (534) cpu_start: Compile time:     Mar 17 2022 11:40:47
I (540) cpu_start: ELF file SHA256:  1a73b8f421cba5f9...
I (546) cpu_start: ESP-IDF:          v5.0-dev-2046-g5963de1caf
I (552) heap_init: Initializing. RAM available for dynamic allocation:
I (559) heap_init: At 3FCA2588 len 0003DA78 (246 KiB): D/IRAM
I (566) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
[... 66 lines cut ...]
0x4200863c: app_main at /Users/andy/esp/esp-rainmaker/examples/led_light/main/app_main.c:139 (discriminator 3)

0x420dcaf9: main_task at /Users/andy/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x40381cbe: vPortTaskWrapper at /Users/andy/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

ELF file SHA256: 1a73b8f421cba5f9

Rebooting...
?ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a40
0x40375a40: esp_restart_noos at /Users/andy/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0110,len:0x18d0
load:0x403b6000,len:0xb70
load:0x403b9000,len:0x319c
entry 0x403b6240

```

shahpiyushv, 2022-03-17:
@eos1d3 , have you checked if the nvs and wifi provisioning examples from esp-idf work fine on your board? Here are the pointers:

https://github.com/espressif/esp-idf/tree/master/examples/storage/nvs_rw_blob

https://github.com/espressif/esp-idf/tree/master/examples/storage/nvs_rw_value

https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr : try reboot after this and check if device reconnects.

eos1d3, 2022-03-18:
For `nvs_rw_blob`, after pressing boot button, the restart counter is always 0.  Log is here:

```
I (33) boot: ESP-IDF v5.0-dev-2046-g5963de1caf 2nd stage bootloader
I (33) boot: compile time 15:27:14
I (33) boot: chip revision: 0
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : SLOW READ
I (47) boot.esp32s3: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0915ch ( 37212) map
I (111) esp_image: segment 1: paddr=00019184 vaddr=3fc90f00 size=02b60h ( 11104) load
I (115) esp_image: segment 2: paddr=0001bcec vaddr=40374000 size=0432ch ( 17196) load
I (124) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1bd70h (114032) map
I (155) esp_image: segment 4: paddr=0003bd98 vaddr=4037832c size=08bcch ( 35788) load
I (165) esp_image: segment 5: paddr=0004496c vaddr=50000000 size=00010h (    16) load
I (170) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
W (185) flash HPM: Enter HPM by reconfiguring dummy has not been fully tested
E (185) flash HPM: Flash high performance mode hasn't been supported
I (189) cpu_start: Pro cpu up.
I (192) cpu_start: Starting app cpu, entry point is 0x40375124
0x40375124: call_start_cpu1 at /Users/andy/esp/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (214) cpu_start: Pro cpu start user code
I (214) cpu_start: cpu freq: 160000000 Hz
I (214) cpu_start: Application information:
I (217) cpu_start: Project name:     nvs_rw_blob
I (222) cpu_start: App version:      v5.0-dev-2046-g5963de1caf
I (229) cpu_start: Compile time:     Mar 18 2022 15:27:06
I (235) cpu_start: ELF file SHA256:  7d24392957dd1ab8...
I (241) cpu_start: ESP-IDF:          v5.0-dev-2046-g5963de1caf
I (248) heap_init: Initializing. RAM available for dynamic allocation:
I (255) heap_init: At 3FC94480 len 0004BB80 (302 KiB): D/IRAM
I (261) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (268) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (274) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (281) spi_flash: detected chip: mxic
W (285) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (296) spi_flash: flash io: dio
W (300) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (314) sleep: Configure to isolate all GPIO pins in sleep state
I (320) sleep: Enable automatic switching of GPIO sleep configuration
I (327) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Restart counter = 0
Run time:
Nothing saved yet!
I (390) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

```

For `nvs_rw_value`, it is the same.  The counter is not increased and I never see "Restart counter = 1".

For `wifi_prov_mgr`, after scanning QR Code, it failed at this screen:

https://i.postimg.cc/tRtVjDN7/2022-03-18-15-52-15.jpg

eos1d3, 2022-03-18:
@shahpiyushv 

Please see https://github.com/espressif/esp-idf/issues/8547#issuecomment-1072302537

My board is ESP32-S3 N32R8V, with a V which uses 1.8V SPI flash, compared to 3.3V SPI in other N8R8 SoC. This can be the compatibility problem. And I suspect the board has some other design issue as the RGB LED is not working as well.

MahendraDnkTech, 2023-01-05:
Hi.. Guys 
Hi.. @shahpiyushv 
I'm also facing a similar issue As my board is **ESP32-S3-DevKitC-1 N32R8V** with 32MB Flash & 8MB of the PSRAM 
For the instance, it's not generating the QRCode for the provisioning 

I'm using a switch example from the rainmaker SDK 
IDF version is V4.4

What will be the solution?

Thanks 
Mahendra