espressif/esp-rainmaker #113: Feature request for reporting OTA status to main application

sahilkhanna, 2022-02-14:
I have a somewhat a complicated application running and with a couple of RTOS Tasks. I found out that OTA fails to work with the whole application running. Sometimes it is SSL handshake that fails or some other reason but from my understanding it is lack of available free heap memory.
I was thinking that it might be good if I could get an event in my main application that OTA has begun so I could suspend/delete all of my tasks of choosing before the system could start downloading the new image

sanketwadekar, 2022-12-23:
Hi @sahilkhanna
We have added OTA Events to Rainmaker. An event handler for these events has also been added in the switch example for your reference.
Moreover, we have also added a new config `ESP_RMAKER_OTA_DISABLE_AUTO_REBOOT` that emits `RMAKER_OTA_EVENT_REQ_FOR_REBOOT` event, allowing you to manually handle reboot after OTA Firmware Image has been downloaded.

shahpiyushv, 2022-12-27:
Closing this issue since the required support has been added.

closed 2022-12-27 by shahpiyushv