espressif/esp-rainmaker #116: Reporting the float value at the app

ankit-thealchemist, 2022-03-02:
Hello, 

I am developing the battery level sensor on rainmaker app. The temperature example is perfect for me. It is compiling fine and working great. The only issue is at the front end, it is showing the degree centigrade.  I just want the simple float value. If I get the % sign, I would be great. How to achieve it?

sahilkhanna, 2022-03-03:
You can create your own param type instead of using temp param type and add it to your device.

```
esp_rmaker_param_t *esp_float_value_param_create(const char *param_name, const char *param_type, float val)
{
    esp_rmaker_param_t *param = esp_rmaker_param_create(param_name, param_type,
            esp_rmaker_float(val), PROP_FLAG_READ | PROP_FLAG_TIME_SERIES);
    if (param) {
        esp_rmaker_param_add_ui_type(param, BOX_RMAKER_UI_TEXT);
    }
    return param;
}

ankit-thealchemist, 2022-03-03:
> You can create your own param type instead of using temp param type and add it to your device.
>
> ```
> esp_rmaker_param_t *esp_float_value_param_create(const char *param_name, const char *param_type, [...]

thanks for making the speedy reply.  Could I also place the '%' sigh on the ui? Can I convert the whole data in to the string and then send eg  like 60%. 

Thanks in advance!

sahilkhanna, 2022-03-03:
> 

I haven't tried it myself but I don't see why that wouldnt work -

```
esp_rmaker_param_t *esp_float_value_param_create(const char *param_name, const char *param_type, const char*val)
{
    esp_rmaker_param_t *param = esp_rmaker_param_create(param_name, param_type,
            esp_rmaker_str(val), PROP_FLAG_READ | PROP_FLAG_TIME_SERIES);
    if (param) {
        esp_rmaker_param_add_ui_type(param, BOX_RMAKER_UI_TEXT);
    }
    return param;
}

```
and then use sprintf to make a string from your percentage value with % in the end and use that to update the param value.
```
esp_rmaker_param_update_and_report(battery_param, esp_rmaker_str(batt_percentage));
```

ankit-thealchemist, 2022-03-03:
Thanks again for the reply!  I appreciate your efforts!
I got every thing about the above snippet. what is the this flag PROP_FLAG_TIME_SERIES ?

Moreover I find resemblance of rainmaker api with the ble mesh api on esp -idf. Is it  a co incidence?

sahilkhanna, 2022-03-03:
> Thanks again for the reply! I appreciate your efforts! I got every thing about the above snippet. what is the this flag PROP_FLAG_TIME_SERIES ?
> 
> Moreover I find resemblance of rainmaker api with the ble mesh api on esp -idf. Is it a co incidence?

oh sorry you dont need that prop to make things work. I just copied the code from one of my examples and modified it to fit your need. 
I dont work or affiliated with espressif but I use their stuff.

sahilkhanna, 2022-03-03:
> 

moreover `PROP_FLAG_TIME_SERIES` is used for time series data which is only meant for float values at the moment so you should disregard that.

shahpiyushv, 2022-03-03:
@ankit-thealchemist 

> Moreover I find resemblance of rainmaker api with the ble mesh api on esp -idf. Is it a co incidence?

That would indeed be a coincidence since different teams have been working on these.

Meanwhile, as @sahilkhanna has mentioned, you can create your own string parameter and add a % sign to the value reported there. Another option is to have the % in your param name itself (Eg. "Battery %", or"Battery percentage") and just continue to use float values. Ofcourse, it is not exactly like what you want, but it would be cleaner and will serve the purpose.

ankit-thealchemist, 2022-03-04:
thanks for all the support has been provided to me.! I am able to get the desired results. 

Thanks again for designing such a great cloud solution.

closed 2022-03-04 by ankit-thealchemist