espressif/esp-rainmaker #117: unable to do the claiming process in esp-idf v4.4

ankit-thealchemist, 2022-03-04:
hello, 

I am able to compile and connect my device to the app when I compile the code using esp-idf v4.3.2. When I try to compile with v4.4 it shows error while claiming the device. I am using the ble as a transport. I havent checkout with the wifi as a transport.  

Actually I have to create the code into the arduino. Since arduino core version >= 2 uses esp idf 4.4, on this also I am unable to claim the device using the ble. 
Is there is any function to claim the device using the wifi? if so please let me know.

shahpiyushv, 2022-03-04:
@ankit-thealchemist , what is the error that you see? Is this with iOS or Android?

ankit-thealchemist, 2022-03-04:
> @ankit-thealchemist , what is the error that you see? Is this with iOS or Android?

Thanks for the speedy reply! Actually it is doing nothing. I am showing the screenshot. My device is redmi note 4. running on android 7.0 NRD90M.  The thing is I have no problem on esp idf v4.3.2. It works perfectly fine 
I would try with different device and let you know about the results.

[image]
[image]
[image]

khushbushah2302, 2022-03-04:
Hi @ankit-thealchemist , 
Can you remove (forget) device from phone's Bluetooth settings and try for provisioning again ? 
Steps to remove deice : Bluetooth Settings -> Check previously connected devices. -> Click on settings button for that device -> Click Forget.

ankit-thealchemist, 2022-03-04:
I checked with other phone and is working. But its strange its not working with my phone. ok I have no issue now! closing the request!

closed 2022-03-04 by ankit-thealchemist