espressif/esp-rainmaker #118: Can't clone from github, always clone fails.

Ocean100, 2022-03-06:
Can't clone from github, always fails to clone, some can succeed, and some have been unable to clone.
For example: fatal: clone of 'https://github.com/espressif/json_parser.git'

The standard commands are used:
git clone --recursive https://github.com/espressif/esp-rainmaker.git

Ocean100, 2022-03-06:
[image]

shahpiyushv, 2022-03-07:
@Ocean100 , I see some gitclone.com paths in your logs. Are you using some caching logic via gitclone.com? The default repository  and submodules do not have gitclone.com URLs, so seems more like an environment issue to me.

Ocean100, 2022-03-07:
@shahpiyushv 
Using gitclone.com is an attempt to speed things up, actually the same result without this.
I tried using git clone --recursive git://github.com/espressif/esp-rainmaker.git
It is found that most files can be cloned normally, and some small files can also be cloned manually according to the prompts.
I don't quite understand the difference between https and git. I am trying to find the reason here.

Ocean100, 2022-03-20:
The final reason: network problem, I have used someone else's computer remotely and can clone successfully.

shahpiyushv, 2022-04-01:
Closing this issue since it has been resolved.