espressif/esp-rainmaker #120: [rainmaker/Alexa problem] Using rainmaker to debug with Alexa in China

Ocean100, 2022-03-20:
I can't use Alexa in China (AWS seems to have discontinued support for this and I can't open the Alexa app), but I'd like to have voice debugging integrated.
What should I do please?

Ocean100, 2022-03-20:
I installed a DualSpace software that can virtualize the phone desktop and download Alexa, but the phone app will eventually report an error

shahpiyushv, 2022-03-21:
@Ocean100 , the problem is not completely clear to me, but if the problem is at Alexa side itself, and if it the commands aren't even hitting the RainMaker backend , it could be hard to debug the issue. What kind of errors does Alexa report?

shixinke-orion, 2022-04-02:
@Ocean100 You need to switch the language to English, turn off the mobile communication network, remove the sim card, connect to the network using WiFi, and finally enable the VPN client (if you have a VPN service).