espressif/esp-rainmaker #122: ESP Rainmaker HomeKit Switch Example Please set HOMEKIT_PATH to esp-homekit-sdk repo

maialbe2823, 2022-04-16:
[image]
I get this error when trying out the ESP Rainmaker HomeKit sample.
However; when I try to declare the HOMEKIT_PATH I get the following error:
[image]
Any ideas on how to resolve this please. I have try different methods but none seems to work out.  It only happens when I try the ESP Homekit SDK. 
Thanks

shahpiyushv, 2022-04-18:
@maialbe2823 how exactly are you setting the HOMEKIT_PATH? As per the example [README](https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch#setup), you need to set it using

```
$ export HOMEKIT_PATH=/path/to/esp-homekit-sdk/
```

This works atleast on MacOS and Linux when used from the terminal.

maialbe2823, 2022-04-18:
Yeah , but how and where do I set that path?

shahpiyushv, 2022-04-22:
@maialbe2823 , as mentioned above, if you are using macOS or Linux terminal, `export HOMEKIT_PATH=/path/to/esp-homekit-sdk/` is what you should execute on the terminal (basically, from the same place where you execute idf.py build). In your [earlier comment](https://github.com/espressif/esp-rainmaker/issues/122#issue-1206008659), you mentioned "However; when I try to declare the HOMEKIT_PATH ...". Where did you declare this?

maialbe2823, 2022-04-22:
I declared it in the CmakeLists of the project since I saw it in one of the examples. So, I’m other words that’s only for Mac or Linux. But I’m doing this in Windows, and it still doesn’t run and it gives me those errors. How can I fix it on windows?

shahpiyushv, 2022-04-22:
@maialbe2823 , you should set HomeKit_PATH at the same place where you have set your IDF_PATH. I believe the steps [here](https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/add-idf_path-to-profile.html) may work, just by replacing IDF_PATH with HOMEKIT_PATH. Meanwhile, setting it in CMakeLists.txt may also work, but you may require to give the complete absolute path like `set(HOMEKIT_PATH "C:/path/to/esp-homekit-sdk")`. Have you tried that?

maialbe2823, 2022-04-24:
I did both ways but got this error instead:
 for bothCMake Error:
  Running

   'C:/Users/.../.espressif/tools/ninja/1.10.2/ninja.exe' '-C' 'C:/Users/maial/OneDrive/Desktop/test/esp-rainmaker/examples/homekit_switch/build' '-t' 'recompact'

  failed with:

   ninja: error: build.ninja:21102: lexing error

shaurya0406, 2022-07-18:
Even I am facing the same issue.
I have modified the Cmakelists.txt file to add path and extra component directories of HomeKit SDK.

`# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# if(NOT DEFINED $ENV{HOMEKIT_PATH})
#     message(FATAL_ERROR "Please set HOMEKIT_PATH to esp-homekit-sdk repo")
# endif(NOT DEFINED ENV{HOMEKIT_PATH})

if(DEFINED ENV{HOMEKIT_PATH})
  set(HOMEKIT_PATH $ENV{HOMEKIT_PATH})
else()
  set(HOMEKIT_PATH ${/Volumes/SanDisk_Extreme/ESP32/esp-homekit-sdk})
endif(DEFINED ENV{HOMEKIT_PATH})

if(DEFINED ENV{RMAKER_PATH})
  set(RMAKER_PATH $ENV{RMAKER_PATH})
else()
  set(RMAKER_PATH ${CMAKE_CURRENT_LIST_DIR}/../..)
endif(DEFINED ENV{RMAKER_PATH})

# Add RainMaker components and other common application components
# set(EXTRA_COMPONENT_DIRS $ENV{HOMEKIT_PATH}/components/homekit ${RMAKER_PATH}/components/esp-insights/components ${RMAKER_PATH}/components ${RMAKER_PATH}/examples/common components)
set(EXTRA_COMPONENT_DIRS ${HOMEKIT_PATH}/components ${HOMEKIT_PATH}/components/homekit ${HOMEKIT_PATH}/examples/common ${RMAKER_PATH}/components/esp-insights/components ${RMAKER_PATH}/components ${RMAKER_PATH}/examples/common)

set(PROJECT_VER "1.0")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(homekit_switch)`

shaurya0406, 2022-07-18:
Even I am facing the same issue.
I have modified the Cmakelists.txt file to add path and extra component directories of HomeKit SDK.

```
cmake_minimum_required(VERSION 3.5)
if(DEFINED ENV{HOMEKIT_PATH})
  set(HOMEKIT_PATH $ENV{HOMEKIT_PATH})
else()
  set(HOMEKIT_PATH ${/Volumes/SanDisk_Extreme/ESP32/esp-homekit-sdk})
endif(DEFINED ENV{HOMEKIT_PATH})

if(DEFINED ENV{RMAKER_PATH})
  set(RMAKER_PATH $ENV{RMAKER_PATH})
else()
  set(RMAKER_PATH ${CMAKE_CURRENT_LIST_DIR}/../..)
endif(DEFINED ENV{RMAKER_PATH})set(EXTRA_COMPONENT_DIRS ${HOMEKIT_PATH}/components ${HOMEKIT_PATH}/components/homekit ${HOMEKIT_PATH}/examples/common ${RMAKER_PATH}/components/esp-insights/components ${RMAKER_PATH}/components ${RMAKER_PATH}/examples/common)

set(PROJECT_VER "1.0")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(homekit_switch)
```