espressif/esp-rainmaker #123: Matter support?

twlawrenceko, 2022-05-11:
Hello,

I have visited the ESP blog regarding to the Matter support on ESP32. It seems like it can be loaded as a CMake library?

Thank you.

shubhamdp, 2022-05-11:
@twlawrenceko are you looking for Matter support on ESP32 or RainMaker + Matter support on ESP32?

We support Matter on ESP32 chips and you can try the [lighting app](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32) example in [connectedhomeip](https://github.com/project-chip/connectedhomeip) repository. There are few more examples in `examples/*/esp32` path.

We do not support RainMaker + Matter in upstream chip repository.

twlawrenceko, 2022-05-11:
@shubhamdp , keep me updated once the Matter support is integrated with RainMaker then I will close this thread.

JackSuen, 2022-12-17:
@shubhamdp Is there any progress on the work RainMaker + Matter?

emangler, 2023-01-04:
I am working on a ESP32 RainMaker + Matter Project. 
Is there any progress on the RainMaker + Matter Example?
Thank You!

tharunlok, 2023-05-18:
is there any update