espressif/esp-rainmaker #124: Pairing RainMaker Assist Claiming after HomeKit Pairing?

twlawrenceko, 2022-05-11:
Hello,

Once the ESP device is paired via HomeKit QRcode (MFi in our case specifically). How do we link into RainMaker app? 

This is the terminal log we see once HomeKit is paired up.
```
E (361034) esp_rmaker_core: Node connected to Wi-Fi without Assisted claiming. Cannot proceed to MQTT connection.
E (361044) esp_rmaker_core: Please update your phone apps and repeat Wi-Fi provisioning with BLE transport.
```

When attempting to scan the RainMaker provisioning QRcode, it shows as device not found.

Thank you

shahpiyushv, 2022-05-12:
@twlawrenceko there are 2 things required for RainMaker to work

1. MQTT credentials
These can be automatically fetched using self claiming for esp32c3, esp32s2 and esp32s3. For esp32, the assisted/host claiming are the only options. Since assisted claiming is linked to BLE provisioning, it cannot be performed after the node is already paired. [Host claiming](https://rainmaker.espressif.com/docs/claiming.html#host-driven-claiming-esp32) is the only option then.

2. User - Node association
Even this happens during provisioning. If some other provisioning is being used, additional efforts are required to perform the user node association, using the underlying [C APIs](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html#_CPPv434esp_rmaker_start_user_node_mappingPcPc). Some of our customers, who have HomeKit + RainMaker have built a custom HomeKit service for RainMaker user association and used that from their HomeKit + RainMaker phone app to perform the association after HomeKit pairing.

twlawrenceko, 2022-05-14:
@shahpiyushv , Thank you for the details.

For the unified provisioning process (that is included in ESP-IDF repo), it seems to only connect ESP to local Wi-Fi network but not pairing to the app or phone. Or in other words, once the device is paired, how would the phone application link back to the ESP that is on the same network?

So for ESP RainMaker we know it will go through similar process as unified provisioning, but what is method of linking the device back to RainMaker app on phone? Is it processed and searching the ESP device locally or through AWS?

Thank you.

twlawrenceko, 2022-05-14:
From some answer [here](https://github.com/espressif/esp-idf-provisioning-android/issues/25#issuecomment-772396331), the softAP provisioning is receiving the success since the softAP is working while STA is also working, which softAP isn't closed yet. But what is the method for future local connection that is done in RainMaker?

shahpiyushv, 2022-05-16:
@twlawrenceko , as you can see [here](https://rainmaker.espressif.com/docs/user-node-mapping.html) the user-node mapping for RainMaker happens during the provisioning itself. The information is sent to the node just before sending the wifi credentials. Post provisioning, there is no way to do the user node association using this method. But as mentioned earlier, for a case like HomeKit provisioning, you can use a custom HomeKit service to do the user-node mapping on local network.