espressif/esp-rainmaker #125: error: implicit declaration of function 'portMUX_INITIALIZE'

jacek12345, 2022-05-24:
I can't compile any of example:
`error: implicit declaration of function 'portMUX_INITIALIZE'; did you mean 'portMUX_NO_TIMEOUT'? [-Werror=implicit-function-declaration]
     portMUX_INITIALIZE(&pxNewRingbuffer->mux);`
Using ESP-IDF 4.4

There is in rbuf.c:
`#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 4, 0)
    portMUX_INITIALIZE(&pxNewRingbuffer->mux);
#else
    vPortCPUInitializeMutex(&pxNewRingbuffer->mux);
#endif`

But i didn't find where the 'portMUX_INITIALIZE' is declared

jacek12345, 2022-05-27:
Can I use vPortCPUInitializeMutex in place of portMUX_INITIALIZE in ESP-idf 4.4? Is it correct? It makes no compilation error.

shubhamdp, 2022-05-27:
@jacek12345 Can you please tell me the rainmaker commit id and esp-idf commit id that you are using.

jacek12345, 2022-05-27:
rainmaker: commit b4db703281af865b59d607821483114adb3e1e3f (HEAD -> master, origin/master, origin/HEAD)
esp-idf: commit 9b46f4e086b656fe3307c3ff92e8c476ccd2d1c6 (HEAD, tag: v4.4-beta1)

shubhamdp, 2022-05-27:
This is fixed in esp-idf stable release v4.4.1. Is it possible for you to switch to stable [tag v4.4.1](https://github.com/espressif/esp-idf/releases/tag/v4.4.1) release?

jacek12345, 2022-05-27:
I tried to switch to 4.4.1, but i have now:
Toolchain:
  C:/Users/jacek/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe,
  crosstool-ng version esp-2021r2 doesn't match supported version
  esp-2021r2-patch3

shubhamdp, 2022-05-27:
re running `install.bat` and `export.bat` should be able to resolve this.

jacek12345, 2022-05-27:
Thank You, i run install.bat but not export.bat. Now my old project, that i was checking after changing to 4.4.1, compiles.
But in rainmaker example now i have:
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/cbor (0.5.4)
[2/2] idf (4.4.1)
CMake Error at F:/GitHub/esp-idf/tools/cmake/component.cmake:238 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "cbor" candidates: "cbor, espressif__cbor"

Call Stack (most recent call first):
  F:/GitHub/esp-idf/tools/cmake/build.cmake:491 (__component_get_requirements)
  F:/GitHub/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:16 (project)

-- Configuring incomplete, errors occurred!
See also "F:/GitHub/esp-rainmaker/examples/switch/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

shubhamdp, 2022-05-27:
There is an issue with including `cbor` component with idf-component-manager. Can you please downgrade the idf-component-manager to v1.0.1 and try again?

```
pip uninstall idf-component-manager
pip install idf-component-manager==1.0.1
```

This issue will be fixed sooner

jacek12345, 2022-05-27:
Thank You so much, now it works

closed 2022-05-27 by jacek12345