espressif/esp-rainmaker #133: MDNS component not found after changes in IDF 5.0

espzav, 2022-07-08:
Please, we are using esp_rainmaker component in esp-bsp example project and this failing now, because they are changes in mdns compnent in IDF 5.0. The mdns component is moved to component manager. Please, could you help me with fix?

Thank you very much.

shahpiyushv, 2022-07-12:
@espzav Can you please create a file at location `components/esp_rainmaker/idf_component.yml` with this content:

```
## IDF Component Manager Manifest File
dependencies:
  espressif/mdns:
    version: "^1.0.3"
    rules:
      - if: "idf_version >=5.0"
```

Please note that release/v5.0 branch is still not stable and so it may have other breaking changes in future.

espzav, 2022-07-12:
Thank you, @shahpiyushv. It's working. May I make a PR for it? 

And I know, that v5.0 is not stable, but these breaking changes is blocking for other development, because there is failing automatic tests.

shahpiyushv, 2022-07-12:
@espzav good to know that this worked for you. We already have an internal MR for this, which will be merged shortly.

espzav, 2022-07-12:
@shahpiyushv Ok, I found it. Thank you very much!

closed 2022-07-12 by espressif-bot

dizcza, 2022-12-10:
8 days since IDF v5.0 is officially available stable version.

I'm getting the same error though on v5.1. I've set up a separate env v4.4 'cause I imagine there are other components like esp-crypto, M5Core2-for-AWS, etc. that don't work on v5.x yet.

Just to let you know that there are users waiting for 5.x support.

shahpiyushv, 2022-12-11:
@dizcza , if your issue is similar to [this issue reported on forum](https://www.esp32.com/viewtopic.php?f=41&t=30201&sid=546bc0229eaa7c283b1b09ace2a0b0f6) , please try `pip install --upgrade idf-component-manager`. If the error is something different, I would recommend filing a new Issue for better tracking.

dizcza, 2022-12-11:
Actually, when I switched to IDF v5.0 branch, I managed to build the "switch" and "temperature_sensor" examples with ease, nice work. I did have issues with v5.1 master branch (I've not copied the build error, but it was similar to the author's post), but no complaints, it's not officially supported yet.

Thanks for supporting v5.0.