espressif/esp-rainmaker #134: Cannot provision new device

homonto, 2022-07-10:
So I am trying ESP32-S2 and when trying to provision I am getting errors on the application (serial monitor shows lots of data):
[image]
[image]
[image]

manuferro, 2022-07-20:
I've the same problem after the updating all at last release (esp-idf, esp-rainmaker, esp-rainmaker-android)
before it was working

shahpiyushv, 2022-07-20:
@homonto , for your issue with esp32s2, I think your phone's Wi-Fi settings may be causing it to disconnect from the esp32s2 as the phone would not be getting Internet on that Wi-Fi interface. Typically, the OS gives a popup asking if you want to stay connected to the Wi-Fi. Do you get that popup?

Meanwhile, I would suggesting using the release/v4.4 branch of esp-idf and ensuring that all submodules are in sync using `git submodule update --init --recursive`. 

For your issue @manuferro , can you double check that the PoP you are providing is correct?

manuferro, 2022-07-21:
Pop is inserted correctly (QR, manually, softAP, BLE). (also if is not recognized correctly)
Switching back on esp-idf 4.4 it's working.

Thansk