espressif/esp-rainmaker #135: building example project missing member in struct

PratikMistry-cbi, 2022-07-12:
Hi. 

I'm trying to build the example switch project of the esp-rainmaker. However receiving the following error.

 error: 'esp_local_ctrl_config_t' {aka 'struct esp_local_ctrl_config'} has no member named 'proto_sec'
         config.proto_sec.version = sec_ver;
               ^
C:/Users/Pratik.Mistry/esp/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_local_ctrl.c:293:15: error: 'esp_local_ctrl_config_t' {aka 'struct esp_local_ctrl_config'} has no member named 'proto_sec'
         config.proto_sec.custom_handle = NULL;
               ^
C:/Users/Pratik.Mistry/esp/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_local_ctrl.c:297:15: error: 'esp_local_ctrl_config_t' {aka 'struct esp_local_ctrl_config'} has no member named 'proto_sec'
         config.proto_sec.pop = pop;
               ^
[1137/1144] Building ASM object esp-idf/esp_rainmaker/CMakeFiles/__idf_esp_rainmaker.dir/__/__/rmaker_claim_service_server.crt.S.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

please assist

shahpiyushv, 2022-07-12:
@PratikMistry-cbi , have you pulled in the latest esp-rainmaker which has some fixes for newer idf versions?

PratikMistry-cbi, 2022-07-13:
@shahpiyushv yes on 2f0f5e0 branch