espressif/esp-rainmaker #136: Cannot generate QR code in led_light example

Parteiadler, 2022-07-14:
Hello,
I’m trying to build the example “led_light” project in esp-rainmaker. When monitoring the output, I received the error below:
I (2022) wifi_prov_mgr: Provisioning started with service name : PROV_1b361c
I (2022) app_wifi: Provisioning started
I (2022) esp_rmaker_local: Event 1
I (2032) app_wifi: Scan this QR code from the ESP RainMaker phone app for Provisioning.

Traceback (most recent call last):
  File "C:\Tools\esp-idf\tools\idf.py", line 808, in <module>
    main()
  File "C:\Tools\esp-idf\tools\idf.py", line 743, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\Users\Parteiadler\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Parteiadler\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\Parteiadler\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\click\core.py", line 1691, in invoke
    return _process_result(rv)
  File "C:\Users\Parteiadler\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\click\core.py", line 1628, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
  File "C:\Users\Parteiadler\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Tools\esp-idf\tools\idf.py", line 638, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "C:\Tools\esp-idf\tools\idf.py", line 216, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "c:\tools\esp-idf\tools\idf_py_actions\serial_ext.py", line 159, in monitor
    RunTool('idf_monitor', monitor_args, args.project_dir, build_dir=args.build_dir, hints=hints)()
  File "C:\Tools\esp-idf\tools\idf_py_actions\tools.py", line 146, in __call__
    process, stderr_output_file, stdout_output_file = asyncio.run(self.run_command(self.args, env_copy))
  File "C:\Program Files\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\Tools\esp-idf\tools\idf_py_actions\tools.py", line 180, in run_command
    await asyncio.gather(
  File "C:\Tools\esp-idf\tools\idf_py_actions\tools.py", line 214, in read_and_write_stream
    output_file.write(output)
UnicodeEncodeError: 'gbk' codec can't encode character '\u2580' in position 3: illegal multibyte sequence
Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x0000023E6DF468B0>
Traceback (most recent call last):
  File "C:\Program Files\Python38\lib\asyncio\base_subprocess.py", line 126, in __del__
  File "C:\Program Files\Python38\lib\asyncio\base_subprocess.py", line 104, in close
  File "C:\Program Files\Python38\lib\asyncio\proactor_events.py", line 108, in close
  File "C:\Program Files\Python38\lib\asyncio\base_events.py", line 719, in call_soon
  File "C:\Program Files\Python38\lib\asyncio\base_events.py", line 508, in _check_closed
RuntimeError: Event loop is closed
Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x0000023E6DF61F70>
Traceback (most recent call last):
  File "C:\Program Files\Python38\lib\asyncio\proactor_events.py", line 116, in __del__
  File "C:\Program Files\Python38\lib\asyncio\proactor_events.py", line 108, in close
  File "C:\Program Files\Python38\lib\asyncio\base_events.py", line 719, in call_soon
  File "C:\Program Files\Python38\lib\asyncio\base_events.py", line 508, in _check_closed
RuntimeError: Event loop is closed

It seems like something went wrong when generating the QR code.
What should I do?

shahpiyushv, 2022-07-18:
@Parteiadler , not really sure why this error has occurred, but seems some issue while rendering the code via python. A workaround would be to disable CONFIG_APP_WIFI_PROV_SHOW_QR so that the QR code is not printed at all and instead, use the URL that is printed on the serial terminal.

shahpiyushv, 2023-06-06:
Closing this issue since it is stale.

closed 2023-06-06 by shahpiyushv