espressif/esp-rainmaker #137: Is there a way to disable user inputs/configs conditionally?

Thomasalex2, 2022-07-19:
Hey all,

First of all, I absolutely love Rainmaker and thanks for making it a reality.

I have a WS2812 project where I would like to disable the hue, saturation and brightness sliders when the user selects a preset colour theme/gradient from the dropdown in the app? Is that currently possible to do?

I am using Rainmaker on Arduino IDE

shahpiyushv, 2022-07-19:
@Thomasalex2 Modifying the visibility of the UI elements or enabling/disabling them at run time is not possible at the moment.

Thomasalex2, 2022-07-22:
It would actually be great if we could organize and create many different settings and parameters through the ESP rainmaker app.

A similar question, is there an easy way to update UI elements from functions without manually invoking the callback function? I'm using Rainmaker on the Arduino IDE

xhivo97, 2022-08-17:
> It would actually be great if we could organize and create many different settings and parameters through the ESP rainmaker app.

I actually have a really good use case for this,  I need to have sort of a setting page for my device where I can:
1. Run calibration
2. Change a variable

The only _clean_ way I can think of is to have a separate view on the settings page of that device. I also would love for that device to keep a log of events that I can view directly on the app and not just on ESP Insights.

Is that even possible without server side changes?

CC-MASTER05, 2023-03-01:
Hi @Thomasalex2 ,

I am using same example as you are using and i have tested with default pin it working fine but once i change pin at a time not working is same happening with you? or can you please help me to change pin?

Regards,
Fenil Ghoghari