espressif/esp-rainmaker #138: Using NimBle in app makes crash after provisioning

jacek12345, 2022-07-20:
I need constantly using BLE in rainmaker.  Communicate with BLE sensor (using NimBle, as RainMaker while BLE provisioning). 
It works since provisioning is already done.
When I need to make provisioning, app crashes on my NimBle initialisation.
I placed   NimBle initialisation after app_wifi_start() function:

` /* Start the Wi-Fi.
     * If the node is provisioned, it will start connection attempts,
     * else, it will start Wi-Fi provisioning. The function will return
     * after a connection has been successfully established
     */
    err = app_wifi_start(POP_TYPE_RANDOM);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Could not start Wifi. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }
    
   //My NimBle initilalisation
    ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());
    nimble_port_init();
    /* Initialize the NimBLE host configuration. */
    ble_hs_cfg.reset_cb = bleprph_on_reset;
    ble_hs_cfg.sync_cb = bleprph_on_sync;
    nimble_port_freertos_init(bleprph_host_task);

`

There is app crash after provisioning and connecting wifi:

`I (1185) app_wifi: Provisioning Started. Name : PROV_89a126, POP : 0aea9731
I (1195) app_wifi: Provisioning will auto stop after 30 minute(s).
I (34205) protocomm_nimble: mtu update event; conn_handle=1 cid=4 mtu=256

I (47645) esp_rmaker_user_mapping: Received request for node details
I (47645) esp_rmaker_user_mapping: Got user_id = SignInWithApple_t11111111111111111111, secret_key = 11111111-1111-1111-1111-111111111111
I (47655) esp_rmaker_user_mapping: Sending status SUCCESS
I (47925) app_wifi: Received Wi-Fi credentials
        SSID     : xxxxxxx
        Password : xxxxxxxxxxxxx
I (48935) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (48935) wifi:state: init -> auth (b0)
I (49635) wifi:state: auth -> assoc (0)
I (49645) wifi:state: assoc -> run (10)
I (49665) wifi:connected with xxxxx, aid = 36, channel 1, BW20, bssid = xx:xx:xx:xx:xx:xx
I (49665) wifi:security: WPA3-SAE, phy: bgn, rssi: -60
I (49735) wifi:pm start, type: 1

I (49735) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (49745) wifi:BcnInt:102400, DTIM:3
W (49855) wifi:<ba-add>idx:0 (ifx:0, xx:xx:xx:xx:xx:xx), tid:0, ssn:1, winSize:64
I (50625) app_wifi: Connected with IP Address:192.168.1.63
I (50625) esp_netif_handlers: sta ip: 192.168.1.63, mask: 255.255.255.0, gw: 192.168.1.1
I (50625) wifi_prov_mgr: STA Got IP
I (50625) app_wifi: Provisioning successful
I (50635) esp_rmaker_local: Waiting for Wi-Fi provisioning to finish.
I (50645) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (50645) esp_mqtt_glue: Waiting for MQTT connection. This may take time.
W (50655) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40389863
0x40389863: _esp_error_check_failed at F:/GitHub/esp-idf/components/esp_system/esp_err.c:42

file: "../main/app_main.c" line 318
func: app_main
expression: esp_nimble_hci_and_controller_init()

abort() was called at PC 0x40389867 on core 0
0x40389867: _esp_error_check_failed at F:/GitHub/esp-idf/components/esp_system/esp_err.c:43

Stack dump detected
Core  0 register dump:
[... 62 lines cut ...]
3fc9fea0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0000
3fc9fee0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: b19322dab83446a1
ELF file SHA256: b19322dab83446a1

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380654
0x40380654: esp_restart_noos_dig at F:/GitHub/esp-idf/components/esp_system/esp_system.c:44 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16bc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000`

I have configured  CONFIG_APP_WIFI_PROV_TRANSPORT_BLE in menuconfig

Is it seems that BLE is not deinitialised after provisioning? And therefor it crashes while my NimBle initialisation?
What should I do if i want to use NimBle in app all the time (besides provisioning time).

jacek12345, 2022-07-21:
Thank You for patch. I have problem with installing it:

F:\GitHub\esp-idf>git apply BT_configurable.patch
error: patch failed: components/wifi_provisioning/Kconfig:39
error: components/wifi_provisioning/Kconfig: patch does not apply

F:\GitHub\esp-idf>

IshaESP, 2022-07-21:
Hi,
Patch is getting applied at my end. Can you try making the changes manually for that file if possible.

Thanks,
Isha

jacek12345, 2022-07-21:
Sorry, i think i'm doing something wrong because no files are modified at my end. But if i understand from Your patch file analysing, NimBle controller should be deinitialised after provisioning without additional flags from Your patch? It only adds some possibility for not disabling if it is needed? Is it correct or I missunderstood?
If that is correct, i don't need Your patch since i want to have NimBle always disabled after provisioning?
So it seems that something is not properly deinitialised after provisioning, when my esp_nimble_hci_and_controller_init() initialisation begins?

jacek12345, 2022-07-21:
I changed all files manually.
SET in menuconfig:
WIFI_PROV_KEEP_BT_ON_AFTER_PROV = y
WIFI_PROV_DISCONNECT_AFTER_PROV = n

Also changed from
`ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());`
to
`ESP_ERROR_CHECK_WITHOUT_ABORT(esp_nimble_hci_and_controller_init());`

BLE works after provisioning, but .... the provisioning hangs on phone app
[image]

IshaESP, 2022-07-21:
Hi @jacek12345,
First of all, my patch was against master branch. I will work on backporting it for release branches.

Basically, I analyzed your changes and I understand the issue now. You are trying to initialize nimble immediately after app_wifi_start() function returns. In normal case, stack deinitialization after provisioning is an asynchronous process. This involves posting message to a thread which is then processed in stack context. For your change, I suggest to add a delay (Example - vTaskDelay(2000);) before attempting to initialize nimble.

This should help resolve your crash issue.

app_wifi_start()
 **vTaskDelay(2000)**
 < nimble initialization> 

Thanks,
Isha

jacek12345, 2022-07-21:
Thank You for response. You right, it is asynchronous proces. I tried with delay but this is much more time needed than 2s. Is any convenient callback function or parameter that i can check if stack ends deinitjalisation (is deinitialised), so could initialise it immediately if provisioning didn't occured or after deinitialisation if provisioning was done?

jacek12345, 2022-07-25:
Thank You Isha, it looks that
vTaskDelay(2000); helps.

But, could You check flow with Your patch with WIFI_PROV_KEEP_BT_ON_AFTER_PROV disabled?
I think, that there is something wrong with deinitialisation.
First, i tried with:
 WIFI_PROV_KEEP_BT_ON_AFTER_PROV = y
 WIFI_PROV_DISCONNECT_AFTER_PROV = y

In monitor, i see that NimBle stack deinitialisation ends with 
'protocomm_nimble: Protocomm layer has already stopped'
and next, there is another initialisation:
`NimBLE: GAP procedure initiated: advertise;
I (35344) NimBLE: disc_mode=2
I (35344) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (35354) NimBLE:`

I expected, that with
WIFI_PROV_KEEP_BT_ON_AFTER_PROV = n
it should simply ends with
'protocomm_nimble: Protocomm layer has already stopped'
 but it doesn't.
`I (42494) NimBLE: GAP procedure initiated: stop advertising.

I (42504) NimBLE: GAP procedure initiated: stop advertising.

I (42514) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

E (42554) protocomm_nimble: Error setting advertisement data; rc = 30
W (42574) Timer: Timer not stopped
W (42584) Timer: Timer not stopped
I (42604) wifi_prov_mgr: Provisioning stopped
W (42604) BTDM_INIT: esp_bt_mem_release not implemented, return OK
I (42604) wifi_prov_scheme_ble: BTDM memory released
I (42604) esp_rmaker_local: Event 6`

Is it correct deinitialisation flow? I can initialise NimBle after that, but I'd like to be sure if it is safe.

IshaESP, 2022-07-27:
Hi @jacek12345 ,
When WIFI_PROV_KEEP_BT_ON_AFTER_PROV is disabled ('n'), that is the expected behaviour because we are stopping the bluetooth host stack here. The clean way to do this it to stop advertising, terminate the connection and then stop the stack. Hence all the console prints.

Thanks,
Isha

closed 2022-07-27 by jacek12345