espressif/esp-rainmaker #139: can anyone say how to upgrate my blynk cloud base code

jamsyogendra, 2022-07-20:
Hello Esp-rainmaker

to be very honest review right now esp rainmaker is the one of best iot platfrom right now i am trying to switch on it
here is my blynk cloud related coding, 

can anyone help me for modify my coding part for esp rainmaker 

```
#include <ESP8266WiFi.h>           // Header file for Wifi Module 
#include <BlynkSimpleEsp8266.h>  // header file for blynk 
#include <RBDdimmer.h>           // header file for robodyn dimmer

char auth[] = "";  // blynk auth key 

char ssid[] = "";   // Wifi Name
char pass[] = "";  // Wifi Password

#define outputPin  D2  // PWM pin for dimmer
#define zerocross  D1 // for boards with CHANGEBLE input pins

dimmerLamp dimmer(outputPin, zerocross); //initialase port for dimmer for ESP8266, ESP32, Arduino due boards

int outVal = 0; // Intialisation value for dimmer
int dim_val; // dimmer value to dim the appliance

void setup()
{
  Serial.begin(9600); // begin serial communication
  dimmer.begin(NORMAL_MODE, ON); //dimmer initialisation: name.begin(MODE, STATE)
  Blynk.begin(auth, ssid, pass); // begin blynk server
}

void loop()
{
  Blynk.run(); // Run blynk server
}

BLYNK_WRITE(V0)  {  // Set to your virtual pin  
  outVal = param.asInt(); // Get State of Virtual Button
  dim_val = map(outVal, 0, 1023, 0, 100); // mapped the value for dimmer
  dimmer.setPower(dim_val);               // Set dimmer power
  Blynk.virtualWrite(V1, dim_val); //sending to Blynk
}
```

shahpiyushv, 2022-07-21:
@jamsyogendra , have you already tried the RainMaker examples in Arduino? if not, please read [this blog post](https://blog.espressif.com/esp-rainmaker-now-in-arduino-cf1474526172). Also, note that if you use ESP RainMaker with ESP IDF, instead of Arduino, you will get much more flexibility. You can get started by following [these docs](https://rainmaker.espressif.com/docs/get-started.html)

jamsyogendra, 2022-07-21:
Yes, today i tried Rainmaker with my Device ( 8 Channel Relays and 1 Zero Cross base Fan Dimmer) with IR Remote Control, Manually Control, Mobile Control , Alexa, Google Home, also getting live feedback of button from everyside almost everything 
but only one problem my dimmer power(ON/OFF) feature is working but i am unble to control brightness using slider 

[image]