espressif/esp-rainmaker #140: ESP32C3 provision crash

FlynnMa, 2022-07-21:
Hi,

I recently created my new workspace using esp rainmaker to run with my esp32c3. I managed to build idf and the 'switch' example from esp-rainmaker, here is the guide I am following:
https://rainmaker.espressif.com/docs/get-started.html

But always, I get crashed on provisioning, I couldn't figure out why on 3 days debugging, here is my log looks like:
```
I (1929) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"jade_42e94e","pop":"75574429","transport":"softap"}
I (1949) app_main: Provisioning QR : {"ver":"v1","name":"jade_42e94e","pop":"75574429","transport":"softap"}
I (1959) app_wifi: Provisioning Started. Name : jade_42e94e, POP : 75574429
I (1969) app_wifi: Provisioning will auto stop after 30 minute(s).
I (26999) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (26999) wifi:station: f2:86:d8:28:06:b0 join, AID=1, bgn, 40U
I (27219) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
W (28169) wifi:<ba-add>idx:2 (ifx:1, f2:86:d8:28:06:b0), tid:0, ssn:15, winSize:64
W (45699) protocomm_httpd: Resetting socket session id as socket 57 was closed
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420b63be  RA      : 0x420b63aa  SP      : 0x3fcb27d0  GP      : 0x3fc91c00  
TP      : 0x3fc73468  T0      : 0x4208a038  T1      : 0x0000000f  T2      : 0xf3bd6f1a  
S0/FP   : 0x00000010  S1      : 0x37353537  A0      : 0x00000002  A1      : 0x37353537  
A2      : 0x00000010  A3      : 0x00000001  A4      : 0x37353537  A5      : 0x6003b080  
A6      : 0x00000000  A7      : 0x7fffffff  S2      : 0x00000002  S3      : 0x00000001  
S4      : 0x00000040  S5      : 0x00000010  S6      : 0x00000004  S7      : 0x3fcb2894  
S8      : 0x00000002  S9      : 0x3fcb3be9  S10     : 0x00000001  S11     : 0x00000000  
T3      : 0x00000008  T4      : 0x465701fe  T5      : 0xef456e96  T6      : 0xb6dbf879  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x37353537  
MHARTID : 0x00000000  

Stack memory:
3fcb27d0: 0x00000001 0x3fcaede4 0x00003434 0x37353537 0x00000002 0x000000d0 0x00000000 0x4208e672
3fcb27f0: 0x00000000 0x00000000 0x37353537 0x37353537 0x00000000 0x00003400 0x00003434 0x4208e8f2
3fcb2810: 0x3fcb3c09 0x3fcb3be9 0x3fcb3c29 0x3fcab140 0x00000000 0x00003400 0x37353537 0x3fcb286c
3fcb2830: 0x00000000 0x00000000 0x00003434 0x4208a104 0x4207bd94 0x3fcb2968 0x3fcb3a9c 0x3fcb2900
3fcb2850: 0x00003434 0x37353537 0x00000000 0x420869e0 0x00000000 0x00000004 0x00000000 0x00003434
3fcb2870: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2890: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb28b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb28d0: 0x00000000 0x00000001 0x00000002 0x00000001 0x3fcab370 0x3fcaae28 0x3fcab070 0x3fcb3be4
3fcb28f0: 0x3fcab370 0x3fcaae28 0x3fcab070 0x4202ba00 0x00000000 0x3fcb299c 0x3fcb2d90 0x42046cce
3fcb2910: 0x00000000 0x42060b34 0x3fcb2958 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcaad24
3fcb2930: 0xdab50537 0x3fcb2a28 0x3fcb2a2c 0x3fcb3a9c 0x3fcaabf0 0xdab50537 0x3fcb3be4 0x4202be16
3fcb2950: 0x00000000 0x00000000 0x3fca0374 0x4202be6a 0x0000004f 0x3fcb29ec 0x3c108780 0x00000000
3fcb2970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000029 0x3fcb3aec 0x3fcb2a28
3fcb2990: 0x3fcb2a2c 0x3fcb12e9 0x3fcae9e8 0x4202a228 0x3fcb2b2c 0x3fcb12e9 0x3fcae9e8 0x4202a56c
3fcb29b0: 0x00000000 0x3fcaad24 0x00000029 0x4206e43a 0x00001800 0x00000029 0x00000000 0x00000001
3fcb29d0: 0x3fcaad24 0x3fcb12e0 0x3fcb1504 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001
3fcb29f0: 0x3fcaad24 0x3fcb12e0 0x00000029 0x4202ae2c 0x73736573 0x3d6e6f69 0x39363633 0x34363932
3fcb2a10: 0x00003933 0x73736573 0x3d6e6f69 0x39363633 0x34363932 0x00003933 0x00000000 0x00000000
3fcb2a30: 0x00000000 0x00000000 0x00000000 0x3fcb12e0 0x3fcb1284 0x3fcb12e0 0x3fcb1284 0x4206edc8
3fcb2a50: 0x3fcb1284 0x00000000 0x00000000 0x00000000 0x3fcb1284 0x3fcb12e0 0x00000000 0x4206da08
3fcb2a70: 0x00000000 0x00000000 0x00000000 0x4206d1a0 0x00000000 0x4206d6ae 0x4206d116 0x4206d392
3fcb2a90: 0x4206d640 0x4206d5b4 0x00000000 0x00000000 0x3fcb12e0 0x00000005 0x00000000 0x00000000
3fcb2ab0: 0x00000000 0x00000001 0x00000028 0x00000132 0x00150608 0x00000000 0x00000000 0x00000000
3fcb2ad0: 0x00010001 0x1f030000 0x3fcb2a78 0x00000000 0x3fcb1504 0x3fcb1760 0x3fcb1284 0x4206dad4
3fcb2af0: 0x00000000 0x00000000 0x00000000 0x3fcb2b5c 0x3fcb1760 0x3fcb1760 0x3fcb1284 0x4206e17a
3fcb2b10: 0x00000000 0x3fcb2b5c 0x3fcb1760 0x4206c5f4 0x00000000 0x4206c5a8 0x3fcb1760 0x4206dc92
3fcb2b30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcb1284 0x4206c8a6
3fcb2b50: 0x00000000 0x00000000 0x00000000 0x3fcb2b68 0x3fcb1284 0x00000039 0x00000000 0x02000000
3fcb2b70: 0x00000000 0x00000000 0x3fcb1284 0x4206c986 0x00000000 0x00000000 0x00000000 0x4038bb6c
3fcb2b90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb2bb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fcb26e0 0x3fc996b4 0x3fc94fdc

ELF file SHA256: 82b766789d5e581f
```

I tried with command:
```
riscv32-esp-elf-addr2line -pfiaC -e build/switch.elf 0x420b80f6
```

here is what I get when I provision with WIFI:
```
0x420b80f6: wpa_ap_rx_eapol at /home/flynn/proj-garden/02-code/esp-idf/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpa_main.c:168

```

I also tried with BLE, it also crashes similarly, the above command would return:
```
0x420b80f6: sha_ll_fill_text_block at /home/flynn/proj-garden/02-code/esp-idf/esp-idf/components/hal/esp32c3/include/hal/sha_ll.h:115
 (inlined by) sha_hal_hash_block at /home/flynn/proj-garden/02-code/esp-idf/esp-idf/components/hal/sha_hal.c:83
```

FlynnMa, 2022-07-21:
update:

If I copy the `provision url` and open from browser, I can successfully scan it, but still I get failed like:

[image]

with logs:
```
I (1675) protocomm_nimble: BLE Host Task Started
I (1685) NimBLE: GAP procedure initiated: stop advertising.

I (1685) NimBLE: GAP procedure initiated: advertise; 
I (1695) NimBLE: disc_mode=2
I (1695) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (1705) NimBLE: 

I (1715) wifi_prov_mgr: Provisioning started with service name : PROV_42e94e 
I (1715) app_wifi: *1 - Provisioning started
I (1725) esp_rmaker_local: Event 1
I (1725) app_wifi: Scan this QR code from the ESP RainMaker phone app for Provisioning.
                                      
  █▀▀▀▀▀█ ▄▄█   ▄█▄ ▄▀█▄▀█  █▀▀▀▀▀█   
  █ ███ █ ▀▄  ▄  ██▀ ▄█  ▄█ █ ███ █   
  █ ▀▀▀ █ █▀█ █▀██ ▄▄▀█▀▀▀ █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ █ █▄▀ ▀ ▀▄▀ █▄▀▄█ ▀▀▀▀�▀▀   
  ██▄█▄ ▀█▄▄ ██▀▄▀█▄█ ▀██▀ ▄█▀▀▄▀▀▄   
  ▄ ▄▄█�▀▀▄▄▄██▄█▄   ▀ ▄▀▄▄ ▄█    █   
  █▀▄▄█▄▀ ▀▄▀█ ▀▀  ██▄ █▄█▄▀█ ▄▄ ▀█   
  ▄ █ ▄█▀█████ █   ▄▀  ▄▄▀█▀  ▄█ █▄   
  ▀▀▄�▀ ▀█▀ ▄ █▄▄ ███▀██  ▄█▀ ▀▀▄ ▀   
  ██▀▄▄▀▀  █▀▀█▀▀ ▀▄█▄▀██ ▄ █▄▄▄ ▀█   
   ▄█ ▄▄▀▀█▄█▄ ▀▀▀█▄▀▀▀▀▀▄█ ▄▄ █ ▄▄   
  ▀█▄   ▀▀ ▄▀  █▄▄▀▀▀▀█▄▀▄▀█  ▄▄ ▀▀   
  ▀ ▀▀▀▀▀▀▄▄ ▄█▀  ▀▀▀▄▀██ █▀▀▀█▀ ▄▄   
  █▀▀▀▀▀█ ▀▀  █▄█ ▀█ ▄ █▀▀ ▀ ██ █    
  █ ███ █ ▄▄▀▀▀▄█ █▀▀▀ ██ ▀████▀ ▄█   
  █ ▀▀� █ ▄  ▄▄▀▀▄▀█▀▄▄▄█ ▀  ▄▀█▀▀▀   
  ▀▀▀▀▀▀▀ ▀ ▀  ▀▀    ▀ ▀  ▀ ▀ ▀▀ ▀    
                                      

I (1915) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_42e94e","pop":"75574429","transport":"ble"}
I (1935) app_wifi: Provisioning Started. Name : PROV_42e94e, POP : 75574429
I (1945) app_wifi: Provisioning will auto stop after 30 minute(s).
I (16385) NimBLE: GAP procedure initiated: advertise; 
I (16385) NimBLE: disc_mode=2
I (16385) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (16395) NimBLE: 
```

Then I continue to manually provision, I got the same crash happen...

sanketwadekar, 2022-09-09:
Hi @FlynnMa,
Do you still have this issue? Which idf version are you currently using? I've tested this using esp32 c3 on idf v4.4.

maxakash, 2022-09-14:
> 

Same here, my esp32 is stuck at confirming wifi connection.

nasimshah05, 2022-09-15:
Need help,esp32 is stuck. tried both scan and manual(ble).
Tried examples and some sketches also from YouTube.still unable to confirm the wifi connection.
[image]

shahpiyushv, 2022-09-15:
The most probable reason for this is that the [wifi_prov_mgr_start_provisioning()](https://github.com/espressif/esp-idf/blob/master/components/wifi_provisioning/include/wifi_provisioning/manager.h#L318) API has changed in release/v5.0. This API may further change, fixing this issue automatically.

nasimshah05, 2022-09-15:
Is there any other way to solve the problem

sanketwadekar, 2022-09-16:
@nasimshah05 Use IDF 4.4 for compiling rainmaker examples