espressif/esp-rainmaker #142: Fan dimmer not working in alexa App

sonetonix-iot, 2022-07-23:
All parameters are listed in alexa App. i can control lights/switches from the alexa App,but i can't able to control dimmer/fan speed using alexa app.

jamsyogendra, 2022-07-23:
hello i am working on same project can you halp me

jamsyogendra, 2022-07-23:
https://github.com/espressif/esp-rainmaker/issues/139#issue-1311268564

sonetonix-iot, 2022-07-23:
> hello i am working on same project can you halp me

how you integrate the IR Remote control with this esp rain maker App

jamsyogendra, 2022-07-23:
using IR lib

jamsyogendra, 2022-07-23:
#include <IRremote.h>

sonetonix-iot, 2022-07-23:
did you control IR function from ESP RAINMAKER App ?

jamsyogendra, 2022-07-23:
yes i did it also with manually feadback with manual switch and ir switch

sonetonix-iot, 2022-07-23:
it means if you press any button in App, the user's custom IR command will illuminate right?

jamsyogendra, 2022-07-23:
yes

sonetonix-iot, 2022-07-23:
Okay thank you

jamsyogendra, 2022-07-23:
whats about fan dimmer which technics are you using from control fan speed or control brigtness of AC. 

Zerocross of relays base?

sonetonix-iot, 2022-07-23:
both techs used ...circuit side no problem

jamsyogendra, 2022-07-23:
can you share some details about zero cross dimming method

sonetonix-iot, 2022-07-23:
what type of bulb led or incandecent

jamsyogendra, 2022-07-23:
ac bulb or we can say ac dimmer 

https://www.tinytronics.nl/shop/en/robotdyn-thyristor-ac-dimmer-3.3-5v

jamsyogendra, 2022-07-23:
Anything solution

sonetonix-iot, 2022-07-25:
there will be an example program, as you mentioned link. it will work

CC-MASTER05, 2023-04-08:
@jamsyogendra are you able to dimming your ac fan if yes can you please share your time sync function with zero-cross detection?