espressif/esp-rainmaker #143: Association refused temporarily, comeback time 205824 mSec

FlynnMa, 2022-07-25:
I very frequently get this kind of failure at startup time, esprecially after flash new binary on the first boot:
```
I (689) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (719) wifi:mode : sta (60:55:f9:79:7e:84)
I (719) wifi:enable tsf
I (729) wifi:new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:1
I (729) wifi:state: init -> auth (b0)
I (759) wifi:state: auth -> assoc (0)
E (759) wifi:Association refused temporarily, comeback time 205824 mSec
```

20 seconds is just too long for my system, how can I fix it?

shahpiyushv, 2022-07-25:
@FlynnMa , this seems some Wi-Fi specific issue. Can you file it in [esp-idf issues](https://github.com/espressif/esp-idf/issues)?

FlynnMa, 2022-07-25:
Hi @shahpiyushv, here it is https://github.com/espressif/esp-idf/issues/9428