espressif/esp-rainmaker #146: CMake error with ESP-IDF v4.4.1

Maldus512, 2022-07-25:
Hello,
I am encountering an error when trying to use esp-rainmaker with ESP-IDF v4.4.1 (git tag), seemingly related to this other issue https://github.com/espressif/esp-rainmaker/issues/133. 

```
$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/maldus/Projects/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=0 /home/maldus/Projects/esp32"...
-- Found Git: /usr/bin/git (found version "2.32.0") 
-- Component directory /home/maldus/Projects/esp-rainmaker/components/esp-insights does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/maldus/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/maldus/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/maldus/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
4 problems were found in the manifest file /home/maldus/Projects/esp-rainmaker/components/esp_rainmaker/idf_component.yml:
Invalid manifest format
Invalid dependency format
Dependency version spec format is invalid
Unknown keys in dependency details: rules
CMake Error at /home/maldus/Source/esp-idf-v4.4.1/tools/cmake/build.cmake:455 (message):
Call Stack (most recent call first):
  /home/maldus/Source/esp-idf-v4.4.1/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:17 (project)

-- Configuring incomplete, errors occurred!
See also "/home/maldus/Projects/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

The error is present even in the provided examples (temperature_sensor) and it seems to be related to the `components/esp_rainmaker/idf_component.yml` file that was recently added. If I remove it the project compiles without issues.

shahpiyushv, 2022-07-26:
@Maldus512 , can you update your idf component manager using `pip install --upgrade idf-component-manager` and retry?

Maldus512, 2022-07-26:
Updating the python package indeed fixed it. Shouldn't it be done when running `install.sh`?

closed 2022-07-26 by Maldus512