espressif/esp-rainmaker #150: How to change node name?

jacek12345, 2022-07-29:
Is it possible to change node name? I didn't find any function for it.
It should be modifiable since name displays on mobile app as main, over node's devices. It should be possible for user to manage it.

shahpiyushv, 2022-07-29:
@jacek12345 we did not add the facility for changing node name because

1. Most uses cases have 1:1 mapping between node and device and so, the device name is sufficient.
2. 3rd party integrations like Alexa/GVA (and even HomeKit for that matter, if that is also added) don't show the concept of node and treat the individual devices under node as independent named entities and so, the node name gets ignored anyways.
3. Since node does not have a dedicated page like devices, adding UI just for changing name becomes a hassle.

The node view for multi-device cases in the RainMaker is just a visual indication that these devices are part of a single node. The node name itself can be used to indicate some info about the node like "2-in-1 Switch", "BLE Gateway", etc. which is fixed, and then, the concept of group and device names can be used to identify actual device.

Even then, if your use case requires the node to have a name, you can create a custom service as per [this blog post](https://blog.espressif.com/extending-esp-rainmaker-with-services-9ccb82dfb9b7), with a "name" parameter which your apps can use to show/edit the node name.

jacek12345, 2022-07-29:
Thank You, I will do that later as custom service.

> The node name itself can be used to indicate some info about the node like "2-in-1 Switch", "BLE Gateway", etc. which is fixed, and then, the concept of group and device names can be used to identify actual device.

In my opinion, this is not so important for end user to displaying node name (that don't identify which one the node is) on top of multidevice group. It should be rather in node details section. It could be displayed user managable name on top of multidevice node. Now this is even hard to choose nodes while creating group since each node have the same name.

There is one more thing about group creating. We should choose devices, but in fact, we can choos whole nodes only.  Is it correct, that we can't get onlsy some devices from multidevice nodes and group it together?

jacek12345, 2023-01-16:
I added additional editable "name" parameter, thank You.

closed 2023-01-16 by jacek12345