espressif/esp-rainmaker #154: Request the mobile app's push notification feature

Andrisetianabrata, 2022-08-05:
Hi, does ESP Rainmaker have a push notification feature?. If so, how i do that? I can't find any examples for push notifications

JohnMacrae, 2022-08-05:
Use this: esp_rmaker_raise_alert("High Voltage Cutoff");

Kind Regards
John

Andrisetianabrata, 2022-08-05:
> Use this: esp_rmaker_raise_alert("High Voltage Cutoff"); Kind Regards John
> […](#)

how to use it the same as arduino.? i used arduino ide to make it

sanket-wadekar, 2022-08-09:
Hi @Andrisetianabrata,
All RainMaker C APIs can work in Arduino too. The ones in Arduino examples are just CPP abstractions on top.

To add push notifications feature to your arduino application, insert `#include <esp_rmaker_core.h>` and call `esp_rmaker_raise_alert("<your-message>");` where you want to send push notifications.

sanketwadekar, 2022-08-09:
Hi @Andrisetianabrata,
All RainMaker C APIs can work in Arduino too. The ones in Arduino examples are just CPP abstractions on top.

To use the push notifications feature, include `esp_rmaker_core.h` file and use `esp_rmaker_raise_alert("<message>");` to send notifications.

shahpiyushv, 2022-09-12:
Closing this issue as appropriate response has been provided.

closed 2022-09-12 by shahpiyushv

JohnMacrae, 2022-10-11:
It works the same in both - try it 😊