espressif/esp-rainmaker #155: ESP Rainmaker disconnects from Wifi after few Hours

ayush9upta, 2022-08-06:
Hi, I've been using ESP Rainmaker for days now, Everything works perfectly except that it gets disconnected after few hours, 
Day 1- 3hrs, 4hrs, 2hrs, 30mins etc..
and same thing happens in the next days
the longest it has stayed connected is 5-6hrs.

The Only solution is to turn off and turn the ESP32 board back on and it connects Instantly again. 
Attaching the code that I'm using below. 
If someone has a solution for this i would be really glad.

[Code_ESP32_RainMaker_IR_DHT11_8Relay_Button.zip](https://github.com/espressif/esp-rainmaker/files/9275542/Code_ESP32_RainMaker_IR_DHT11_8Relay_Button.zip)

jacek12345, 2022-08-09:
Some logs from monitor would be helpfull ...

ayush9upta, 2022-08-20:
I'm using ESP32 Devkit V1
Attaching the Logs:
14:42:07.895 -> LDR - 10.00
14:42:07.989 -> Failed to read from DHT sensor!
14:42:09.966 -> LDR - 10.00
14:42:10.061 -> Failed to read from DHT sensor!
14:42:12.043 -> LDR - 10.00
14:42:12.137 -> Failed to read from DHT sensor!
14:42:14.098 -> LDR - 10.00
14:42:14.193 -> Failed to read from DHT sensor!
14:42:16.179 -> LDR - 10.00
14:42:16.272 -> Failed to read from DHT sensor!
14:42:18.259 -> LDR - 10.00
14:42:18.305 -> Failed to read from DHT sensor!
14:42:20.050 -> E (7620457) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 48
14:42:20.099 -> E (7620457) MQTT_CLIENT: Poll read error: 88, aborting connection
14:42:48.610 -> E (7649023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:42:48.657 -> E (7649024) esp-tls: Failed to open new connection
14:42:48.657 -> E (7649028) TRANSPORT_BASE: Failed to open a new connection
14:42:48.657 -> E (7649033) MQTT_CLIENT: Error transport connect
14:42:48.657 -> E (7649038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:43:17.625 -> E (7678023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:43:17.672 -> E (7678023) esp-tls: Failed to open new connection
14:43:17.672 -> E (7678028) TRANSPORT_BASE: Failed to open a new connection
14:43:17.672 -> E (7678033) MQTT_CLIENT: Error transport connect
14:43:17.672 -> E (7678038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:43:46.643 -> E (7707023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:43:46.643 -> E (7707023) esp-tls: Failed to open new connection
14:43:46.643 -> E (7707028) TRANSPORT_BASE: Failed to open a new connection
14:43:46.691 -> E (7707033) MQTT_CLIENT: Error transport connect
14:43:46.691 -> E (7707038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:44:15.686 -> E (7736023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:44:15.686 -> E (7736023) esp-tls: Failed to open new connection
14:44:15.686 -> E (7736028) TRANSPORT_BASE: Failed to open a new connection
14:44:15.686 -> E (7736033) MQTT_CLIENT: Error transport connect
14:44:15.686 -> E (7736038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:44:44.666 -> E (7765023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:44:44.712 -> E (7765023) esp-tls: Failed to open new connection
14:44:44.712 -> E (7765028) TRANSPORT_BASE: Failed to open a new connection
14:44:44.712 -> E (7765033) MQTT_CLIENT: Error transport connect
14:44:44.712 -> E (7765038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:45:13.679 -> E (7794023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:45:13.679 -> E (7794023) esp-tls: Failed to open new connection
14:45:13.679 -> E (7794028) TRANSPORT_BASE: Failed to open a new connection
14:45:13.726 -> E (7794033) MQTT_CLIENT: Error transport connect
14:45:13.726 -> E (7794038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:45:42.701 -> E (7823023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:45:42.701 -> E (7823023) esp-tls: Failed to open new connection
14:45:42.701 -> E (7823028) TRANSPORT_BASE: Failed to open a new connection
14:45:42.701 -> E (7823033) MQTT_CLIENT: Error transport connect
14:45:42.701 -> E (7823038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:46:11.676 -> E (7852023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:46:11.723 -> E (7852023) esp-tls: Failed to open new connection
14:46:11.723 -> E (7852028) TRANSPORT_BASE: Failed to open a new connection
14:46:11.723 -> E (7852033) MQTT_CLIENT: Error transport connect
14:46:11.723 -> E (7852038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:46:40.706 -> E (7881023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:46:40.706 -> E (7881023) esp-tls: Failed to open new connection
14:46:40.706 -> E (7881028) TRANSPORT_BASE: Failed to open a new connection
14:46:40.706 -> E (7881033) MQTT_CLIENT: Error transport connect
14:46:40.706 -> E (7881038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:47:09.680 -> E (7910023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:47:09.730 -> E (7910023) esp-tls: Failed to open new connection
14:47:09.730 -> E (7910028) TRANSPORT_BASE: Failed to open a new connection
14:47:09.730 -> E (7910033) MQTT_CLIENT: Error transport connect
14:47:09.730 -> E (7910038) esp_mqtt_glue: MQTT_EVENT_ERROR
14:47:38.684 -> E (7939023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
14:47:38.731 -> E (7939023) esp-tls: Failed to open new connection
14:47:38.731 -> E (7939028) TRANSPORT_BASE: Failed to open a new connection
14:47:38.731 -> E (7939033) MQTT_CLIENT: Error transport connect
[... 174 lines cut ...]
15:04:33.754 -> E (8954033) MQTT_CLIENT: Error transport connect
15:04:33.754 -> E (8954038) esp_mqtt_glue: MQTT_EVENT_ERROR
[... cut ...]
15:06:29.756 -> E (9070023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
15:06:29.756 -> E (9070023) esp-tls: Failed to open new connection
15:06:29.756 -> E (9070028) TRANSPORT_BASE: Failed to open a new connection
15:06:29.756 -> E (9070033) MQTT_CLIENT: Error transport connect
15:06:29.756 -> E (9070038) esp_mqtt_glue: MQTT_EVENT_ERROR
15:06:58.757 -> E (9099023) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0

Please Help me solve this @jacek12345

shahpiyushv, 2022-08-22:
@ayush9upta this seems like an error with your network, probably Internet getting lost. Can you double check that your Internet is reliable and the host a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com is reachable?

ayush9upta, 2022-08-27:
@shahpiyushv thanks for your giving time on this issue, unfortunately I’ve tried diagnose my network, I have even provided fixed IP address to the ESP32 and fixed channel for 2.4ghz network, still it disconnects after few hours. If i turn off my network and turn it on, the ESP32 doesn’t reconnect automatically. So the problem is either the code, which doesnt have a reconnect function or I’ve to make other changes to my router which I don’t know

shahpiyushv, 2022-08-29:
@ayush9upta reconnection attempts are definitely being made, as you can see in the logs itself. However, it is not able to resolve the hostname, which typically points to network/DNS issues.

shahpiyushv, 2022-08-29:
@ayush9upta another thing worth checking is the memory consumption. You can add the below code in your project and call the function periodically to track memory consumption

```
#include <esp_heap_caps.h>
static void print_heap(void)
{
    Serial.printf("\tDescription\tInternal\tSPIRAM\n");
    Serial.printf("Current Free Memory\t%d\t\t%d\n",
           heap_caps_get_free_size(MALLOC_CAP_8BIT) - heap_caps_get_free_size(MALLOC_CAP_SPIRAM),
           heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
    Serial.printf("Largest Free Block\t%d\t\t%d\n",
           heap_caps_get_largest_free_block(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
           heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
    Serial.printf("Min. Ever Free Size\t%d\t\t%d\n",
           heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
           heap_caps_get_minimum_free_size(MALLOC_CAP_SPIRAM));
}
```

Tonvina, 2022-10-26:
> Hi, I've been using ESP Rainmaker for days now, Everything works perfectly except that it gets disconnected after few hours, Day 1- 3hrs, 4hrs, 2hrs, 30mins etc.. and same thing happens in the next [...]

> Hi, I've been using ESP Rainmaker for days now, Everything works perfectly except that it gets disconnected after few hours, Day 1- 3hrs, 4hrs, 2hrs, 30mins etc.. and same thing happens in the next [...]

you can add below line of code to    void sysProvEvent(arduino_event_t *sys_event) 

    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
      Serial.println("\nDisconnected. Connecting to the AP again... ");
      WiFi.begin(name wifi, pass);    
      break;