espressif/esp-rainmaker #157: Problem with delete device from node

jacek12345, 2022-08-09:
I try to delete device from node:
```
    err = esp_rmaker_node_remove_device(node, sensor_device[sensnum]);
    if(!err)
    {
        err = esp_rmaker_device_delete(sensor_device[sensnum]);
        if(!err)
        {
            dev_map[sensnum]=0xFF;
            sensor_device[sensnum]=NULL; 
        }
        else
            ESP_LOGE(TAG, "esp_rmaker_device_delete() failed: %u", err);

    }
    else
        ESP_LOGE(TAG, "esp_rmaker_node_remove_device() failed: %u", err);
```
After that it deletes from iOS app (need to refresh), but ONLY if there is local control enabled. If it is disabled in menuconfig, device remains in mobile app.
The second problem, even when local control enabled and device dissapeared, it backs to app on each refresh (when "Reachable on WLAN" message dissapear) and stay there for the moment when message comeback.

https://user-images.githubusercontent.com/34368502/183582250-d8d6f97d-ab43-45cf-b283-1d0111b3bb1b.MP4

shahpiyushv, 2022-08-09:
After adding or removing a device or params, please call the [esp_rmaker_report_node_details() API](https://github.com/espressif/esp-rainmaker/blob/5c81ecf755432ebeffaadc966b67799ea2f7c9d3/components/esp_rainmaker/include/esp_rmaker_core.h#L885) to report updated data to cloud.

jacek12345, 2022-08-09:
Thank You for help, it works now.

closed 2022-08-09 by jacek12345