espressif/esp-rainmaker #158: Can we add two nodes in ESP Rainmaker App.

SubhodipRoy161101, 2022-08-09:
I have created A node with 4 Switch for my Living Room. I want a exact Similar Node for my Drawing Room also. If get a new ESP32 and implement the same code to it and use same account in rainmaker app, will it add a new node in the app or is there anything we need to change in the code so that it identifies it as a new device?

jacek12345, 2022-08-10:
Yes, we can add two nodes in app. No change in code required for second node. Node registration is quite complex but based on generated random key.

shahpiyushv, 2022-08-17:
@SubhodipRoy161101 , is this resolved now?

closed 2022-09-12 by shahpiyushv

MadanMeelan, 2022-12-21:
How many devices can we add in Rainmaker app for free, I need 8 devices for complete home, can we add all 8 devices through public platform? Please help me on this