espressif/esp-rainmaker #159: Problem with parameter confusion

jacek12345, 2022-08-10:
Reproduced on multi_device example.
Using ESP-IDF v4.4.2
ESP32-C3
iOS app
Problem is that edited value of parameter A sometimes is comming to device as value of parameter B. 
You can see this issue on multi_device example. Go to Temperature sensor device and change name to something that is accepted as float (e.g. 25). It is comming as Temperature value, not name value. Sometimes it comes as should. If change name value to other text - it correctly comes as name parameter or sometimes give "Failure! Please enter a valid float value" message on app, that means it treat parameter as temperature, not name value.

https://user-images.githubusercontent.com/34368502/183907889-59cdf50e-efd4-4ce9-8019-899564e6a85d.MP4

From monitor:
```
I (58223) esp_rmaker_param: Received params: {"Temperature Sensor":{"Temperature":14}}
I (60473) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":25.50000}}
I (71113) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
I (72133) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (74203) esp_rmaker_param: Received params: {"Temperature Sensor":{"Name":"abc"}}
I (74203) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Name":"abc"}}
I (81343) esp_rmaker_param: Received params: {"Temperature Sensor":{"Temperature":88}}
```

jacek12345, 2022-08-29:
can You confirm this issue?

shahpiyushv, 2022-08-29:
@jacek12345 , yes we have confirmed this issue and will work on fixing this. This is not related to the backend though.