espressif/esp-rainmaker #160: Rainmaker doesnt sync after power cycle

Mri007, 2022-08-15:
The esp32 doesnt sync with cloud after powering off and turning back on.

sanketwadekar, 2022-08-15:
Please provide more details about the issue. It will also help if you provide logs.

Mri007, 2022-08-15:
Suppose i have one light turned on from mobile app. It remain on till i turn off esp32. After re-powering esp the light remain off..
It doesnt sync with cloud after connecting to wifi i.e. turn the light on again.

shahpiyushv, 2022-08-16:
@Mri007 , we do not have logic of fetching previous state from cloud because it is not guaranteed that the cloud has the latest state. Instead, there is an option to store it in nvs memory. You just need to add the `PROP_FLAG_PERSIST ` as shown [here](https://github.com/espressif/esp-rainmaker/blob/master/components/esp_rainmaker/src/standard_types/esp_rmaker_standard_params.c#L20). Then, after a reboot, the RainMaker core code will invoke your callbacks with the values from nvs. Note that the write callbaqck should be registered for the device before adding the param, for this to work.

shahpiyushv, 2022-09-12:
Closing this issue as appropriate solution has been provided.

closed 2022-09-12 by shahpiyushv