espressif/esp-rainmaker #161: OTA via dashboard and arduino IDE

Michiel-FMI-ImProvia, 2022-08-22:
Is it possible to do the OTA updates via the https://dashboard.insights.espressif.com when using the Arduino IDE? 
up until now I get an Error when using RMaker.enableOTA(OTA_USING_TOPICS); and the ESP reboots

JohnMacrae, 2022-08-22:
I raised it in 2021 here( https://github.com/espressif/arduino-esp32/issues/5420) It has been studiously ignored despite prodding

Michiel-FMI-ImProvia, 2022-08-22:
Yeah, I saw your item. It would be amazing if it was possible to upload via the dashboard and all the devices get the new firmware. Another option is to try it via the ESP-IDF but then I need time to learn how this works. It's just that there is some OTA code in the arduino examples, and it bugs me that it seems to have no functionality.

shahpiyushv, 2022-08-22:
Low heap memory seems to be the root cause for this. Since Arduino has pre compiled libraries which are common across all esp32 specific modules, there are quite some things pulled in even when not used actively. Getting to a stage wherein the RAM usage will be reduced, but without affecting any other functionality/examples will take some time.

JohnMacrae, 2022-08-22:
It works fine in idf.

Michiel-FMI-ImProvia, 2022-08-22:
> Low heap memory seems to be the root cause for this. Since Arduino has pre compiled libraries which are common across all esp32 specific modules, there are quite some things pulled in even when not [...]

I can imagine that this is not a simple problem. Maybe mention this in the Arduino examples to not get people's hopes up.
In short  if I want to use Rainmaker OTA it's smarter to learn the ESP-IDF language.

JohnMacrae, 2022-09-06:
There is a workaround @Michiel-FMI-ImProvia . Use Arduino OTA https://youtu.be/V-g45CFSFr4

Michiel-FMI-ImProvia, 2022-09-06:
Oh I am so exited!! Will watch that video with intense attention!!!

JohnMacrae, 2022-09-06:
I hope this will give some impetus to the Espressif team. If the Arduino
OTA works the surely it's not a heap collision?

me-chiel, 2022-09-06:
exactly....

sanketwadekar, 2022-09-09:
Hi @JohnMacrae,
We are aware of this crash and are working on the fixes. We will fix this issue in the upcoming release of ESP32 Arduino.
Regards,
Sanket

Michiel-FMI-ImProvia, 2022-09-09:
> Hi @JohnMacrae, We are aware of this crash and are working on the fixes. We will fix this issue in the upcoming release of ESP32 Arduino. Regards, Sanket

any idea when this release will happen? This functionality sounds so amazing.

sanketwadekar, 2022-12-23:
Hello,
Arduino 2.0.6 has been released, which fixes the OTA crash issues seen previously. If you face low memory errors like 
`esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00`, please reboot your device once. Doing a reboot after successful provisioning and then starting OTA is recommended.
Thanks!