espressif/esp-rainmaker #163: Enable time series data

CodorStelian, 2022-08-30:
How to properly enable time series data? Is it enabled on server side? because I get "No chart data" on Android app.
I have already added the time series flag on temperature param.

JohnMacrae, 2022-08-30:
Can you point me at the documentation for this please - I will check it out

miroslavpetrov, 2022-08-30:
@CodorStelian Adding the flag is not sufficient to get ts data wokring. The rainmaker library is unfinished and I was given a patch which adds the ts data support on the firmware side. We are using rainmaker on a private aws instance and its working fine. Find attached the patch.
[rmaker patch.zip](https://github.com/espressif/esp-rainmaker/files/9451739/rmaker.patch.zip)

CodorStelian, 2022-08-30:
@miroslavpetrov I thought the rainmaker library is finished and ready to use for this feature.
Thanks, I will use the patch!

shahpiyushv, 2022-08-30:
@CodorStelian , firmware support was not yet pushed out because of the feature missing in the phone apps. We will push it out in a couple of days.

closed 2022-08-31 by CodorStelian

mcmchris, 2022-09-13:
How can I add the PATCH? would be very valuable for me any help on this!

shahpiyushv, 2022-09-15:
The support has now been pushed to github (https://github.com/espressif/esp-rainmaker/commit/b63b9d66f1967afab2cc31ed92dd84a0ad1adea7)