espressif/esp-rainmaker #166: Adding new parameter after esp_rmaker_start()

jacek12345, 2022-09-09:
What is the proper way for adding new parameter for device "on-the-fly"?
I need to add parameter and want see this this param in mobile app.

shahpiyushv, 2022-09-09:
@jacek12345 If you add a new parameter, you can call the `esp_rmaker_report_node_details()` API to report the new config and param value to the RainMaker backend. However, you will need to refresh the phone app for the new parameter to reflect.

jacek12345, 2022-09-09:
1. esp_rmaker_param_create()
2. esp_rmaker_param_add_ui_type()
3. esp_rmaker_device_add_param()
4. esp_rmaker_param_update_and_report()
5. esp_rmaker_report_node_details()
Is it good way or something wrong/unnecessary?

shahpiyushv, 2022-09-09:
Step 4 is not required, since `esp_rmaker_report_node_details()` will report all params. Rest all is fine.

closed 2022-09-09 by jacek12345