espressif/esp-rainmaker #168: Not able to add Internal Blinds along with standard devices

maxakash, 2022-09-15:
I am trying to add Internal Blinds as a device along with other standard devices such as Fan and LightBulb. But  I am not able to do so. 
First InternalBlind type is not found, and if I try to add it via 
```curtain = esp_rmaker_device_create(ESP_RMAKER_DEVICE_BLINDS_INTERNAL, NULL, NULL);
    esp_rmaker_device_add_param(curtain, esp_rmaker_param_create("name", NULL, esp_rmaker_str("Curtain"),
            PROP_FLAG_READ | PROP_FLAG_WRITE | PROP_FLAG_PERSIST));
    esp_rmaker_param_t *openClose_param = esp_rmaker_param_create("direction", NULL, esp_rmaker_int(0), PROP_FLAG_READ | PROP_FLAG_WRITE);
    esp_rmaker_param_add_ui_type(openClose_param, ESP_RMAKER_UI_DROPDOWN);
    esp_rmaker_device_add_param(curtain, openClose_param);
    esp_rmaker_device_assign_primary_param(curtain, openClose_param);
    esp_rmaker_device_add_cb(curtain, write_cb, NULL);
    esp_rmaker_node_add_device(node, curtain);
``` 
 I get "Could not initialise node" error.
Please help me how to add Internal Blinds.
Thanks

sanketwadekar, 2022-09-15:
Hi @maxakash,
Can you provide more logs? If possible, please share the source file so that I can help you better.

maxakash, 2022-09-15:
Hi @sanketwadekar 

Here is my code
```

#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

const char *service_name = "livingRoom";
const char *pop = "maxakash";

// define the Device Names
char deviceName_1[] = "Fan";
char deviceName_2[] = "Living Room Cove Light";
char deviceName_3[] = "Living Room Chandelier";
char deviceName_4[] = "Dining Room Chandelier";
char deviceName_5[] = "Kitchen Chandelier";
char deviceName_6[] = "Outside Cove Light";
char deviceName_7[] = "Curtain";

// define the GPIO connected with Relays and switches
static uint8_t RelayPin1 = 23;  //D23
static uint8_t RelayPin2 = 22;  //D22
static uint8_t RelayPin3 = 21;  //D21
static uint8_t RelayPin4 = 19;  //D19
static uint8_t RelayPin5 = 18;  //D18
static uint8_t RelayPin6 = 5;   //D5
static uint8_t RelayPin7 = 25;  //D25
static uint8_t RelayPin8 = 26;  //D26

static uint8_t wifiLed    = 2;   //D2
static uint8_t gpio_reset = 0;

/* Variable for reading pin status*/
// Relay State
bool toggleState_1 = LOW; //Define integer to remember the toggle state for relay 1
bool toggleState_2 = LOW; //Define integer to remember the toggle state for relay 2
bool toggleState_3 = LOW; //Define integer to remember the toggle state for relay 3
bool toggleState_4 = LOW; //Define integer to remember the toggle state for relay 4
bool toggleState_5 = LOW; //Define integer to remember the toggle state for relay 5
bool toggleState_6 = LOW; //Define integer to remember the toggle state for relay 6
bool toggleState_7 = LOW; //Define integer to remember the toggle state for relay 7
bool toggleState_8 = LOW; //Define integer to remember the toggle state for relay 8

//The framework provides some standard device types like switch, lightbulb, fan, temperature sensor.
static Fan fan(deviceName_1, &RelayPin1);
static LightBulb lightBulb1(deviceName_2, &RelayPin2);
static LightBulb lightBulb2(deviceName_3, &RelayPin3);
static LightBulb lightBulb3(deviceName_4, &RelayPin4);
static LightBulb lightBulb4(deviceName_5, &RelayPin5);
static LightBulb lightBulb5(deviceName_6, &RelayPin6);
static LightBulb lightBulb6(deviceName_7, &RelayPin7);

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {      
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
        printQR(service_name, pop, "ble");
#else
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
#endif        
        break;
        case ARDUINO_EVENT_WIFI_STA_CONNECTED:
        Serial.printf("\nConnected to Wi-Fi!\n");
        digitalWrite(wifiLed, true);
        break;
[... 189 lines cut ...]
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        }
    }
    delay(100);

    if (WiFi.status() != WL_CONNECTED)
    {
      //Serial.println("WiFi Not Connected");
      digitalWrite(wifiLed, false);
    }
    else
    {
      //Serial.println("WiFi Connected");
      digitalWrite(wifiLed, true);
    }

}
```

I am trying to add blinds to this code.

maxakash, 2022-09-15:
> There is no need to use Rainmaker C APIs (Rainmaker using IDF) in Arduino. To add a new device (other than the standard devices), you can use the APIs provided by Arduino Rainmaker.
> 
> ## Example snippet
> ```
> 
> static Device internalBlinds;
> void setup() {
>    // Code
>     internalBlinds = Device("Internal Blind");
>     Param deviceParam("XYZ Param", NULL, value(20), PROP_FLAG_READ | PROP_FLAG_WRITE | PROP_FLAG_PERSIST);
>     deviceParam.addUIType(ESP_RMAKER_UI_TEXT);
>     internalBlinds.addParam(deviceParam); 
> 
>     my_node.addDevice(internalBlinds); // my_node is the 
>    // Code
> }
> ```
> 
> You can read more about the APIs [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/RainMaker)
> 
> If this solves your problem, you can close this issue.

Thanks @sanketwadekar . This is what I was looking for.

maxakash, 2022-09-15:
Hey @sanketwadekar, I have set up my blinds using ```static Device curtain("Blinds", ESP_RMAKER_DEVICE_BLINDS_INTERNAL, NULL);```
But still the blinds are not shown in Google Home. I just checked that ```openPercent``` is a required trait for blinds. 
I think it is not implemented in GVA due to which it is not shown in Google Home. Can you please check?

maxakash, 2022-09-15:
> @maxakash <https://github.com/maxakash> Have you tried adding
> ESP_RMAKER_PARAM_BLINDS_POSITION (esp.param.blinds-position) parameter?

Yes, I have tried it.

maxakash, 2022-09-15:
> @maxakash Have you tried adding ESP_RMAKER_PARAM_BLINDS_POSITION (esp.param.blinds-position) parameter?

@shahpiyushv it worked after adding ESP_RMAKER_PARAM_BLINDS_POSITION. I was using it the wrong way. Thanks for your help.

maxakash, 2022-09-16:
Hey @shahpiyushv , is there any way to implement Google actions blinds attribute ```discreteOnlyOpenClose```, so that the blinds just have open and close function?

shahpiyushv, 2022-09-16:
@maxakash , currently it is hard-coded with a value of false, so that we support the superset of the valid commands. Bit if there is some hardware limitation, we will allow setting this too. Can you file a separate issue for that with appropriate title?