espressif/esp-rainmaker #169: HTTP Get Time Series Data?

mcmchris, 2022-09-15:
I've tried the new Time Series Data support released today, it works fine in the Android APP, but I am working on creating a custom APP for a final product using Rain Maker but I can't put to work the API.

I tried with this: https://swaggerapis.rainmaker.espressif.com/#/Time%20Series%20Data/GetTSData but I always receive an empty response:

[image]

I need help on this, please!

Thanks.

vikas-chandra-mnnit, 2022-09-16:
Hi @mcmchris , Have you provided the query parameters start_time & end_time in your request URL? Information about these parameters is present in the swagger.

mcmchris, 2022-09-16:
Hi @vikas-chandra-mnnit , I've tried everything, and nothing works, maybe If you can send me a working example, I could know what parameters to fix, I also tried naming the param of the device: Temperature, temperature, temperature_sensor, etc and nothing. I think the correct name is "Temperature" with capital T. I am using the temp sensor example code.

shahpiyushv, 2022-09-16:
@mcmchris , please use "DeviceName.ParamName". Eg. If you device name is "Sensor" and parameter name is "Temperature", use "Sensor.Temperature"

mcmchris, 2022-09-16:
Finally!!!! Thanks, this definitely should be in the Docs and the Swagger instructions, many thanks @shahpiyushv !