espressif/esp-rainmaker #171: ESP Rainmaker Android

Halabdelkrim, 2022-09-18:
Hello,

I'm trying to use the ESP Rainmaker example as a core of my project, but i'm a very beginner in developing Android apps with Java language.

I want to add more functionalities to the app, but i'm confused how to modify the ESP Rainmaker android app to send a custom message (JSON format) to the ESP32.

Which function/method should i use to send custom messages to the ESP32?

It would be so appreciated if you can provide me with an example of it, or at least a starting point from where i can go.

Thank you.

shahpiyushv, 2022-09-19:
@Halabdelkrim , what kind of custom data do you need to pass to the device? You can have custom boolean/integer/float/string parameter which you can control from the phone apps without requiring any change.

Halabdelkrim, 2022-09-19:
Thank you for your response.

I want to send a custom string.

Could you please guide me to the right method to use or if there is any example how to make it?

Halabdelkrim, 2022-09-19:
I'm developing a device that make an action based on a predefined temperature and humidity value. This why i want to make a little changes to app, so the users can control it using esp rainmaker app.

shahpiyushv, 2022-09-19:
@Halabdelkrim , you can see [here](https://github.com/espressif/esp-rainmaker/blob/master/examples/gpio/main/app_main.c#L75) an example of a custom boolean parameter. Similarly, you can create a string parameter using something like this

```
esp_rmaker_param_t *str_param = esp_rmaker_param_create("MyString", NULL, esp_rmaker_str("my value"), PROP_FLAG_READ | PROP_FLAG_WRITE);
```

Halabdelkrim, 2022-09-20:
Hi,
Everything now works well.
Thank you so much.

closed 2022-09-20 by Halabdelkrim