espressif/esp-rainmaker #173: ESP Rainmaker "implicit declaration error"

Halabdelkrim, 2022-09-21:
Hello,

I'm playing around the ESP Rainmaker examples, but when i tried to use "esp_rmaker_schedule_get_params()" method, i always face an error "implicit declaration of function", even if I'm including all the necessary headers.

`../main/app_main.c:99:25: error: implicit declaration of function 'esp_rmaker_schedule_get_params'; did you mean 'esp_rmaker_schedule_enable'? [-Werror=implicit-function-declaration]`

It would be so appreciated if  you could help me figure out this issue!

Thank you.

jacek12345, 2022-09-21:
wher Did You get "esp_rmaker_schedule_get_params()" ? I can't see this in documentation:
https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html?highlight=esp_rmaker_schedule#_CPPv426esp_rmaker_schedule_enablev

Halabdelkrim, 2022-09-21:
@jacek12345  Yes i didn't find it in the documentation too, but you can find it in "esp_rmaker_schedule.c" in the components folder.

The reason why i want to use this method is to store the schedules, so they can still works when the esp goes offline.

If you know any other solution to achieve that, please let me know.

Thank you!

shahpiyushv, 2022-09-21:
@Halabdelkrim , you just need to call the [esp_rmaker_schedule_enable()](https://rainmaker.espressif.com/docs/scheduling.html#firmware) API. Rest all is managed by the phone apps and the esp rainmaker core. The same write callback that is registered for write params gets invoked in the context of scheduling too. You can find more information [here](https://rainmaker.espressif.com/docs/scheduling.html).

Halabdelkrim, 2022-09-22:
@shahpiyushv Thank you so much for the answer.

I tried this two ways (inside write callback), but with no result:

`else if (strcmp(esp_rmaker_param_get_name(param), "Schedule") == 0)
    {
        ESP_LOGI(TAG, "Received Schedule = %s for %s - %s",
                 val.val.s, esp_rmaker_device_get_name(device),
                 esp_rmaker_param_get_name(param));
                 printf("Triggered /n");
    }`

`else if (strcmp(esp_rmaker_param_get_name(param), ESP_RMAKER_DEF_SCHEDULE_NAME) == 0)
    {
        ESP_LOGI(TAG, "Received Schedule = %s for %s - %s",
                 val.val.s, esp_rmaker_device_get_name(device),
                 esp_rmaker_param_get_name(param));
                 printf("Triggered /n");
    }`

Also tried "esp_rmaker_param_get_type()" method and unfortunately no result.

Is this the right way to do it? If no, please provide me with an example.

Thank you.

shahpiyushv, 2022-09-22:
@Halabdelkrim , you would not get any callback for the schedule service or the schedule params themselves. The scheduling framework will invoke your devices specific callback at the scheduled time.

Eg. if you have scheduling enabled and from the phone app you set a schedule for setting "Power" to "true" for your Switch at 6pm everyday, then the esp rainmaker core will invoke your Switch devices's callback at 6 pm everyday. A print like [this](https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_main.c#L41) in your callback will also tell you that your callback was invoked because of a schedule "Schedule".

Note that the firmware manages this locally based on the SNTP based time and the timezone configured. It need not be connected to the Internet at that moment.

It would help if you try the examples like switch/led_light and see how adding schedules from phone apps work for them.

Halabdelkrim, 2022-09-22:
@shahpiyushv Thank you sir for the answer.

I use the ESP32 with DS1302 RTC, so there is no way to store the schedules offline and make them works with the time coming from the RTC ?

Thank you.

shahpiyushv, 2022-09-23:
@Halabdelkrim , we use SNTP to sync time, but if you have an external RTC, and you want to use that to set the time, you can just call the `settimeofday()` API with appropriate input. Rest all would be managed automatically.

closed 2022-09-26 by Halabdelkrim