espressif/esp-rainmaker #174: Device works with Alexa app but can't be used as trigger in routine creation in Alexa app

jacek12345, 2022-09-21:
I Have Alexa app on my iPhone and ESP Rinmaker skill enabled. I see my devices and can control (switching or viewing devices state). But when i try to create routine and don't want use command voice for "When it happen" step, but want use state from my smart home device (e.g. if input state in device changed to ON, it will do something...), i'm choosing "Smart Home" from "When this happens" list and there is not anyone from my devices. I see "No devices found". Is it some special restrictions for device that can be use as trigger in alexa routine creation? I tried with some different types of parameter, but basically use  esp.param.mode type. Is it problem with ESP Rainmaker skill or i have to use some specjal parameter in device, or something else?

shahpiyushv, 2022-09-21:
@jacek12345 , I could not see any such option in the Android Alexa app, so I could not validate if it is not showing any device at all, or not showing just the RainMaker devices. I will check internally, but can you

1. Check if other Smart Home devices show up in the list (in case you have any, and can share the skill name they are linked to).
2. Share screenshot of the Alexa app page where you can create this route.

jacek12345, 2022-09-21:
https://user-images.githubusercontent.com/34368502/191615273-7705cd8c-e872-4676-812e-7a6f6732507a.MOV

jacek12345, 2022-09-26:
I have "Shelly Plus 1" switch and doesn't appear on list also.

shahpiyushv, 2022-09-27:
@jacek12345 , Even I found that option in Android (was looking at wrong place earlier). I have several devices from Tuya and Xiaomi, but I couldn't see any of those in the list

jacek12345, 2022-09-28:
I found this on Alexa developers forum:
https://amazon.developer.forums.answerhub.com/questions/226814/smart-home-skill-as-trigger-for-a-routine.html

jacek12345, 2022-09-29:
I created ESP_RMAKER_DEVICE_LOCK device with esp.param.lockstate parameter type (bool). Now can see device as routine creation trigger but it is signed as unresponsive in Alexa app. Device works with RainMaker app. I suppose, that it is skill depended issue?
Created also second device that is ESP_RMAKER_DEVICE_SWITCH type and also added esp.param.lockstate parameter type (bool). It is also available from routine creation trigger list, also unresponsive.
Addidtionally, there is great feature. The Alexa app creates voice password for control ESP_RMAKER_DEVICE_LOCK  device and don't do it for ESP_RMAKER_DEVICE_SWITCH. Great function for security maintaining (changing device type can change security level). But why devices ar unresponsive....? I can't see any action on monitor console when pushing switch on alexa app, so suspect some problem with skill incompatibility. I didn't found "esp.param.lockstate" in esp_rmaker_standard_types.h so probably also not supported in skill?
Can You check?

https://user-images.githubusercontent.com/34368502/192955874-46114650-7777-4356-ba32-9f26f03b4fcb.MP4

jacek12345, 2022-10-04:
Updating from Allexa Team Support (according to devices that can be trigger for routines):

"We do not have official document for supported interfaces but these below are supported.
In the parentheses, I put supported properties. Although some of them might be supported by only Amazon devices.
Please let us know if these interfaces are not working as triggers, we will make sure it with internal team.
* Motion Sensor - https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-motionsensor.html (Motion detected, Motion not detected for x minutes)
* Contact Sensor - https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-contactsensor.html (Contact detected, Contact not detected)
* Doorbell Event - https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-doorbelleventsource.html (Doorbell pressed)
* Security Panel - https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-securitypanelcontroller.html (Security Panel changes state to 'Armed Away', 'Armed Stay', or 'Disarmed'
* Lock - https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-lockcontroller.html (When lock changes state to 'Locked', 'Unlocked' or 'Jammed') No other interfaces are currently supported by Routines as triggers"

Can You confirm if RainMaker is compatible with some of these Alexa interfaces?

shahpiyushv, 2022-10-04:
@jacek12345 , thanks for this info. I will check with the team and provide an answer soon

sinameshksar, 2022-10-16:
Hello Any update on this issue? we are having the same problem. need to define a trigger from rainmaker in the Alexa routine but nothing is available in the Alexa list. Should we expect to be added though an update?

shahpiyushv, 2022-10-17:
We are checking this internally and will provide an update soon. We already support sending events to Alexa asynchronously, so hopefully, only some minor tweaks in the skill may be required.

jacek12345, 2022-10-17:
Thats great! 
While You're working with this, please try to add "Contact Sensor" device as it is strongly needed for us.
Contact Sensor - https://developer.amazon.com/en-US/docs/alexa/device-apis/alexa-contactsensor.html (Contact detected, Contact not detected)

sinameshksar, 2022-10-17:
> We are checking this internally and will provide an update soon. We already support sending events to Alexa asynchronously, so hopefully, only some minor tweaks in the skill may be required.

Thank you. looking forward to hearing about the updates.

dddeepakdhiman, 2022-10-30:
I also need trigger type of device. Following this issue for update

rushwind, 2022-10-30:
Looks like a skill needs to be created to make the rainmaker devices work with routines. I have been looking into it but it would take me a long time to get there. Would be nice if the rainmaker skill was adapted to incorporate all the sensors and items it supports to work with routines. Is this something that is in the works?

shahpiyushv, 2022-11-08:
@rushwind , we are checking what changes are required to support the routines and then will also add support for the missing sensors.

rushwind, 2022-11-08:
That is wonderful news. Thank you and I am looking forward to it. Will you post updates here or somewhere else?
Thanks again!

sinameshksar, 2022-11-14:
> @rushwind , we are checking what changes are required to support the routines and then will also add support for the missing sensors.

Any estimation on how long it might take till this feature adds to the main stream?

jacek12345, 2022-12-19:
@shahpiyushv Do You have any new information?

shahpiyushv, 2022-12-20:
Routines can now be used with the ESP RainMaker Alexa skill for "Locks". We have also updated the standard types docs [here](https://rainmaker.espressif.com/docs/standard-types.html) with information about what params are required for each device type. A lock device (esp.device.lock) with an integer parameter for lock state (esp.param.lockstate) will work with Alexa routines.

jacek12345, 2022-12-22:
@shahpiyushv Thank You for information. Now i can use lock device as trigger. There is some problem, that Alexa don't offer "Unlock" function as an Action for routine, there is only "Lock" function available, but this is Alexa bug i think. I asked Alexa support.
Can You tell when other device types (especially Contact sensor) will be added in RainMaker?

BTW: esp.param.lockstate definition is still missing in standard param types

jacek12345, 2023-01-09:
@shahpiyushv We are close to private deploy and production phase. But need develop last thing, contact sensor device. Can You estimate when this device will be added to RainMaker?

shahpiyushv, 2023-01-11:
@jacek12345 , I have asked to prioritise this. Now that there are no technical unknowns, this should not be hard.

VarunKumaran, 2023-01-19:
> @shahpiyushv Thank You for information. Now i can use lock device as trigger. There is some problem, that Alexa doesn't offer "Unlock" function as an Action for routine, there is only "Lock" function available, but this is Alexa bug i think. I asked Alexa support. Can You tell when other device types (especially Contact sensor) will be added in RainMaker?
> 
> BTW: esp.param.lockstate definition is still missing in standard param types

Hiii @jacek12345 I am also looking for a way to trigger alexa routines from esp rainmaker.. Can you please send me your example code for "Lock" soo that I can test and see... I tried to code myself but unfortunately I had no luck.

jacek12345, 2023-01-20:
@VarunKumaran this should work:

```
esp_rmaker_device_t*    device;
esp_rmaker_param_t*    state_param;
uint8_t state = 0; //for esp.param.lockstate it can be 0,1 or 2 [look](https://rainmaker.espressif.com/docs/standard-types.html#:~:text=N/A-,Lock%20State,-esp.param.lockstate)
device = esp_rmaker_device_create("device name", ESP_RMAKER_DEVICE_LOCK, NULL]);
if(device)
{
     //add editable device name
      esp_rmaker_device_add_param(device,  esp_rmaker_name_param_create("Name", device name));
      //add state param
      state_param = esp_rmaker_param_create("state", "esp.param.lockstate", esp_rmaker_int(state), PROP_FLAG_READ | PROP_FLAG_WRITE);
    if(state_param)
    {
          esp_rmaker_param_add_ui_type(state_param, ESP_RMAKER_UI_SLIDER);
          esp_rmaker_param_add_bounds(state_param, esp_rmaker_int(0), esp_rmaker_int(2), esp_rmaker_int(1));
          esp_rmaker_device_add_param(device, state_param);
          esp_rmaker_device_assign_primary_param(device, state_param);
    }
}
```

VarunKumaran, 2023-01-21:
> @VarunKumaran this should work:
> 
> ```
> esp_rmaker_device_t*    device;
> esp_rmaker_param_t*    state_param;
> uint8_t state = 0; //for esp.param.lockstate it can be 0,1 or 2 
> device = esp_rmaker_device_create("device name", ESP_RMAKER_DEVICE_LOCK, NULL);
> if(device)
> {
>      //add editable device name
>       esp_rmaker_device_add_param(device,  esp_rmaker_name_param_create("Name", "device name"));
>       //add state param
>       state_param = esp_rmaker_param_create("state", "esp.param.lockstate", esp_rmaker_int(state), PROP_FLAG_READ | PROP_FLAG_WRITE);
>     if(state_param)
>     {
>           esp_rmaker_param_add_ui_type(state_param, ESP_RMAKER_UI_SLIDER);
>           esp_rmaker_param_add_bounds(state_param, esp_rmaker_int(0), esp_rmaker_int(2), esp_rmaker_int(1));
>           esp_rmaker_device_add_param(device, state_param);
>           esp_rmaker_device_assign_primary_param(device, state_param);
>           esp_rmaker_device_add_cb(device, write_cb, NULL);     // You should also add appriopriate code in write_cb()
>           esp_rmaker_node_add_device(node, device);
>     }
> }
> ```

Hiii @jacek12345 Thank you for replying... I tried to add your code but I am getting many errors while compiling it's cuz I have no idea on how to add your code in existing esp rainmaker examples.. Can you please send me a full working code.. in which the lock shld be unlocked and locked in loop for every 30sec.. to check and test alexa routines

jacek12345, 2023-01-23:
@VarunKumaran Take the "gpio" example and paste my code after:
```
esp_rmaker_node_t *node = esp_rmaker_node_init(&rainmaker_cfg, "ESP RainMaker Device", "GPIO-Device");
    if (!node) {
        ESP_LOGE(TAG, "Could not initialise node. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }
```
Add `#include <esp_rmaker_standard_params.h>` on the file beginning.
Replace write_cb function with:
```
/* Callback to handle commands received from the RainMaker cloud */
static esp_err_t write_cb(const esp_rmaker_device_t *device, const esp_rmaker_param_t *param,
            const esp_rmaker_param_val_t val, void *priv_data, esp_rmaker_write_ctx_t *ctx)
{
    if (ctx) {
        ESP_LOGI(TAG, "Received write request via : %s", esp_rmaker_device_cb_src_to_str(ctx->src));
    }
    if (app_driver_set_gpio(esp_rmaker_param_get_name(param), val.val.b) == ESP_OK) {
        esp_rmaker_param_update_and_report(param, val);
    }
    else if ((strcmp(esp_rmaker_param_get_type(param), "esp.param.lockstate") == 0))
        esp_rmaker_param_update_and_report(param, val);

    return ESP_OK;
}
```
Now You should detect lock device in Your Alexa app and use it as trigger source in Alexa routine.

VarunKumaran, 2023-01-24:
> Take the "gpio" example and paste my code after:

Can you Please share the GPIO example INO sketch.(I'm trying to write code in arduino ide)

jacek12345, 2023-01-25:
Sorry, i don't use arduino. I suppose that You can get any arduino example and change device and param type and go on.

sanketwadekar, 2023-01-25:
@VarunKumaran 
For Arduino, you can start with the RmakerSwitch example sketch file and modify it to add your devices and parameters for Lock. Please read the detailed API docs for Arduino [here](https://github.com/espressif/esp-rainmaker/issues/174). For the complete list of Alexa supported devices and parameters, refer [this link](https://rainmaker.espressif.com/docs/standard-types.html#devices).

VarunKumaran, 2023-01-25:
> you can start with the RmakerSwitch example sketch file and modify it to add your devices and parameters for Lock

Hiii @sanketwadekar Thank you for replying... I tried to modify the RmakerSwitch Example for lock changing the device name and parameters but I get error while compiling... can you help me with what I have done wrong?? 

``//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

//GPIO for push button
#if CONFIG_IDF_TARGET_ESP32C3
static int gpio_0 = 9;
static int gpio_switch = 7;
#else
//GPIO for virtual device
static int gpio_0 = 0;
static int gpio_switch = 16;
#endif

/* Variable for reading pin status*/
int switch_state = 0;

//The framework provides some standard device types like switch, lightbulb, fan, temperaturesensor.
static Lock my_switch;

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32S2
            Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
            printQR(service_name, pop, "softap");
#else
            Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
            printQR(service_name, pop, "ble");
#endif
            break;
        case ARDUINO_EVENT_PROV_INIT:
            wifi_prov_mgr_disable_auto_stop(10000);
            break;
        case ARDUINO_EVENT_PROV_CRED_SUCCESS:
            wifi_prov_mgr_stop_provisioning();
            break;
        default:;
    }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();

    if(strcmp(param_name, "Lock State") == 0) {
        Serial.printf("Received value = %s for %s - %s\n", val.val.b? "true" : "false", device_name, param_name);
        switch_state = val.val.b;
        (switch_state == 0) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
        param->updateAndReport(val);
    }
}

void setup()
{
    Serial.begin(115200);
    pinMode(gpio_0, INPUT);
    pinMode(gpio_switch, OUTPUT);
    digitalWrite(gpio_switch, DEFAULT_POWER_MODE);

    Node my_node;
    my_node = RMaker.initNode("ESP RainMaker Node");

    //Initialize switch device
    my_switch = Lock("Lock", &gpio_switch);

    //Standard switch device
    my_switch.addCb(write_callback);

    //Add switch device to the node
    my_node.addDevice(my_switch);

    //This is optional
    RMaker.enableOTA(OTA_USING_TOPICS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone().
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();

    RMaker.enableSchedule();

    RMaker.enableScenes();

    RMaker.start();

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32S2
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{
    if(digitalRead(gpio_0) == LOW) { //Push button pressed

        // Key debounce handling
        delay(100);
        int startTime = millis();
        while(digitalRead(gpio_0) == LOW) delay(50);
        int endTime = millis();

        if ((endTime - startTime) > 10000) {
          // If key pressed for more than 10secs, reset all
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        } else {
          // Toggle device state
          switch_state = !switch_state;
          Serial.printf("Toggle State to %s.\n", switch_state ? "true" : "false");
          my_switch.updateAndReportParam(ESP_RMAKER_DEF_LOCKSTATE_NAME, switch_state);
          (switch_state == 0) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
      }
    }
    delay(100);
}``

JohnMacrae, 2023-01-25:
@VarunKumaran  If you're trying to learn ESP Rainmaker, I have a number of videos and code examples you might want to check out: https://www.youtube.com/playlist?list=PL4Um4E3Og2S8It9Dyhdvf3ScvuTA_Tp-i

VarunKumaran, 2023-01-25:
> @VarunKumaran If you're trying to learn ESP Rainmaker, I have a number of videos and code examples you might want to check out: https://www.youtube.com/playlist?list=PL4Um4E3Og2S8It9Dyhdvf3ScvuTA_Tp-i

Hiii @JohnMacrae seems like you are an expert in Rmaker... saw your videos still I coudnt able to find what errors I have done in my code... would you mind helping me with the code?

JohnMacrae, 2023-01-25:
This isn't the place for such discussion. More a place for discussing bugs/improvements with Rainmaker. There are lots of places to seek _specific_ help. Reddit, youtube channels, Facebook pages and the like are full of people seeking and solving problems. But the problems have to be **specific** - and the error must be documented to get a sensible answer. Its a big ask to get people to download your code and compile it for example. They don't have time but they can give you pointers when you ask **specific** questions or pose **specific** problems

tbbusby, 2023-02-15:
> I created ESP_RMAKER_DEVICE_LOCK device with esp.param.lockstate parameter type (bool). Now can see device as routine creation trigger but it is signed as unresponsive in Alexa app. Device works [...]

Hi there, I'm a very new developer for fun... and I'm building an alexa skill that can trigger a routine.. i should say trying but i cannot for the life of me figure out why my contact sensor cannot be chosen to trigger a routine. When i link it, it shows on my account but it isn't selectable from the routine triggers part of the app.  Any advice would be appreciated. below is my index.js

'use strict';
console.log("test");

import AWS from 'aws-sdk';
import crypto from 'crypto';
import { v4 as uuid } from 'uuid';
import AlexaResponse from './alexa/skills/smarthome/AlexaResponse.js';
AWS.config.update({region:'us-east-1'});

//exports.handler = async function (event, context) {

export const handler = async function (event, context) {
    // Dump the request for logging - check the CloudWatch logs

    console.log("index.handler request  -----");
    console.log(JSON.stringify(event));

    if (context !== undefined) {
        console.log("index.handler context  -----");
        console.log(JSON.stringify(context));
    }

    // Validate we have an Alexa directive

    if (!('directive' in event)) {
        let aer = new AlexaResponse(
            {
                "name": "ErrorResponse",
                "payload": {
                    "type": "INVALID_DIRECTIVE",
                    "message": "Missing key: directive, Is request a valid Alexa directive?"
                }
            });
        return sendResponse(aer.get());
    }

    // Check the payload version

    if (event.directive.header.payloadVersion !== "3") {
        let aer = new AlexaResponse(
            {
                "name": "ErrorResponse",
                "payload": {
                    "type": "INTERNAL_ERROR",
                    "message": "This skill only supports Smart Home API version 3"
                }
            });
        return sendResponse(aer.get())
    }

    let namespace = ((event.directive || {}).header || {}).namespace;

    if (namespace.toLowerCase() === 'alexa.authorization') {
        let aar = new AlexaResponse({ "namespace": "Alexa.Authorization", "name": "AcceptGrant.Response", });
        return sendResponse(aar.get());
    }

    // ATTEMPTING TO IMPLEMENT STATE REPORTING

    if (namespace.toLowerCase() === 'alexa.statereport') {
        let uuid = crypto.randomUUID();
        let correlationToken = ((event.directive.header.correlationToken))
   //     let asr = new AlexaResponse({ "namespace": "Alexa", "name": "StateReport", "messageId": uuid, "correlationToken": correlationToken, "payloadVersion": "3" });
        let asr = new AlexaResponse(event)
        asr.addContextProperty({
            "namespace": "AlexaContactSensor", "name": "detectionState", "value": "NOT_DETECTED", "timeOfSample": "timevariable", "uncertaintyInMilliseconds": 0
[... 128 lines cut ...]

function sendDeviceState(endpoint_id, state, value) {
    let dynamodb = new AWS.DynamoDB({apiVersion: '2012-08-10'});

    let key = state + "Value";
    let attribute_obj = {};
    attribute_obj[key] = {"Action": "PUT", "Value": {"S": value}};

    let request = dynamodb.updateItem(
        {
            TableName: "SampleSmartHome",
            Key: {"ItemId": {"S": endpoint_id}},
            AttributeUpdates: attribute_obj,
            ReturnValues: "UPDATED_NEW"
        });

    console.log("index.sendDeviceState request -----");
    console.log(request);

    let response = request.send();

    console.log("index.sendDeviceState response -----");
    console.log(response);
    return true;
}

jacek12345, 2023-02-18:
> @jacek12345 , I have asked to prioritise this. Now that there are no technical unknowns, this should not be hard.

@shahpiyushv Do You have any new information?

shahpiyushv, 2023-02-27:
New devices that support routines have now been added to Alexa skill. Please refer the [standard types](https://rainmaker.espressif.com/docs/standard-types.html) docs for details.

jacek12345, 2023-02-28:
Thank You @shahpiyushv 
I think, there is one issue - Alexa can detect device only after provisioning. No way to discover device if we delete it in Alexa app and trying discover one more time. We have to provision (e.g. scan QR code) one more time. Is it something skill dependent?

shahpiyushv, 2023-02-28:
@jacek12345 , If you do a discovery form the Alexa App itself, it should again sync the device list. Did that not work for you?

jacek12345, 2023-02-28:
Yes, not work. Only work after reset wifi and provisioning.

jacek12345, 2023-03-01:
> New devices that support routines have now been added to Alexa skill. Please refer the [standard types](https://rainmaker.espressif.com/docs/standard-types.html) docs for details.

Do You know which devices supports routines? It would be good to mark it in docs if You know.
E.g. switch does not support it (i don't know if it is skill issue or Alexa limit).

rushwind, 2024-03-07:
I have used the lock device to create an Alexa routine. Thank you. I would like to know which other devices will work with Alexa to create routines. In  https://rainmaker.espressif.com/docs/standard-types.html standard types it does not indicate which ones work with Alexa routines. Kind of a pain to create each device to find out what ones work. If you could post a list of which devices work with routines I would greatly appreciate it.
Thank you

rushwind, 2024-03-07:
Not to be greedy but it would be nice if all devices worked with Alexa routines.