espressif/esp-rainmaker #175: No verification mail received

Rasmus-Fink, 2022-09-26:
Hi,
Using https://dashboard.rainmaker.espressif.com/sign-up I  enter email address and password - but I receive no verification email (checked spam, obviously).
This means that I cannot get started with it.

Any suggestions?

//Fink

fink-at-trmc-dk, 2022-09-27:
Hi again,
Just tried via the app, but same issue. I have obviously checked both spelling and spam folder :-(
Please report back if you need any further information from me :-)

//Fink

shahpiyushv, 2022-09-27:
@fink-at-trmc-dk , can you send an email to esp-rainmaker-admin@espressif.com with the information of email address being used? We can check internally.

JohnMacrae, 2022-09-27:
This has also happened to me in the past. In the end, I used a Google login
which worked fine for the same email address