espressif/esp-rainmaker #178: Device not found

ausafmomin, 2022-10-01:
Esp32 wroom dev kit v1 having firmware installed ,while scanning qr code error pops device not found

sanketwadekar, 2022-10-01:
Hi @ausafmomin 
You can try provisioning manually. If the problem still doesn't get solved, please provide logs here.

ausafmomin, 2022-10-01:
> Hi @ausafmomin You can try provisioning manually. If the problem still doesn't get solved, please provide logs here.

I did tried manual provisioning and 9 out of 10 times its not showing in the app but when i tried with bluetooth terminal app it showed the device and when tried to connect it says GATT STATUS 113 error

jacek12345, 2022-10-04:
what is Your device that You have installed RainMaker app and using for scanning ?

jacek12345, 2022-10-05:
asked about smartphone (Android, iOS), what version?

techiesms, 2023-03-12:
The solution of this and many other Rainmaker problems is shown in this video. watch it out 
https://youtu.be/pNJBaWZ_vj0

jorgb19, 2023-12-09:
No compatible device in your esp rainmaker account