espressif/esp-rainmaker #180: Using SSL certs to publish to API server

chaudhariatul, 2022-10-03:
Hi, 

I'm trying to create a function which allows to publish to data to an API server which uses SSL, so I tried using WiFiClientSecure.h with rainmaker. This allows to set setCACert, setCertificate and setPrivateKey, which enables connecting to the API server. However the **Confirming Node Association** step fails while I keep getting data on my API server.

Any suggestions on how to use SSL and allow publishing data to rainmaker as well as another API server?

shahpiyushv, 2022-10-03:
@chaudhariatul , if you are talking about using the RainMaker REST APIs from the device, then that would not work using the private key and certificate. REST APIs use the AWS Cognito User Authentication and are meant for clients like phone apps and dashboard. This requires username+password which are used to login and get the refresh/access tokens, to be eventually used for the REST APIs.

chaudhariatul, 2022-10-03:
Thanks @shahpiyushv for reply. I'm try to connect and send data to different self hosted server in the cloud. We are using different SSL certificates here. We want to display temperature on rainmaker mobile app as well as send data to the API server.

chaudhariatul, 2022-10-03:
Thanks @shahpiyushv  for reply. I'm try to connect and send data to different self hosted server in the cloud. We are using different SSL certificates here. We want to display temperature on rainmaker mobile app as well as send data to another API server.

shahpiyushv, 2022-10-03:
@chaudhariatul , so if I understand correctly, this is totally unrelated to RainMaker server itself, right? In that case, if you are seeing failures in "Confirming Node Association", it could be an issue of the firmware crashing after it was provisioned. Can you check in your logs and verify?

chaudhariatul, 2022-10-03:
@shahpiyushv I see this in the Serial console:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
Client connection begin.

E (2630) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7680
E (2630) esp-tls: Failed to open new connection
E (2631) TRANSPORT_BASE: Failed to open a new connection
E (2636) HTTP_CLIENT: Connection failed, sock < 0
E (2639) esp_claim: Failed to open connection to https://esp-claiming.rainmaker.espressif.com/claim/initiate
E (2649) esp_claim: Claim Init Request Failed.
E (2652) esp_claim: Claim Init Sequence Failed.
E (2657) esp_rmaker_core: esp_rmaker_self_claim_perform() returned -1. Aborting
```

I can detect ESP32 as bluetooth device and then set the WiFi crendentials. Board is able to connect to WiFi and starts sending data to my API server. On the rainmaker mobile app it keeps progressing through _Configuring Node association_ and then fails at **Confirming Node association**.

Error message as show in the box above is right after I provide the WiFi credentials.

shahpiyushv, 2022-10-03:
Can you try this once

1. Just reboot the device and see if you still see the esp_claim error messages
2. If you are using any default example, press and hold the boot button for more than 3 seconds and release, so that it goes back into provisioning mode.
3. Try provisioning again.
 
**Do not erase/reflash in between**

chaudhariatul, 2022-10-03:
Hi @shahpiyushv I tried these steps and got same error.

I used the default example as a reference but here is my code if it helps to reproduce the issues

```
//Based on Rainmaker example code in Arduino.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"
#include <wifi_provisioning/manager.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>
#include "secrets.h"

#include "Adafruit_SHTC3.h"

#include <WiFiClientSecure.h>
WiFiClientSecure net = WiFiClientSecure();
PubSubClient client(API_ENDPOINT, 8883, net);

unsigned long lastMillis = 0;

static int buffer = 128000;

#define DEFAULT_I2C_PORT &Wire

Adafruit_SHTC3 shtc3 = Adafruit_SHTC3();

// Set Defalt Values
#define DEFAULT_Temperature 0
#define DEFAULT_Humidity 0

//GPIO for push button
#if CONFIG_IDF_TARGET_ESP32C3
static int gpio_0 = 9;
static int gpio_switch = 7;
#else
//GPIO for virtual device
static int gpio_0 = 0;
static int gpio_switch = 16;
#endif

#if CONFIG_IDF_TARGET_ESP32C3
//GPIO for push button
static int gpio_reset = 9;
//GPIO for virtual device
static int gpio_power = 7;
static int gpio_swing = 3;
static int gpio_mode_auto = 4;
static int gpio_mode_cool = 5;
static int gpio_mode_heat = 6;
static int gpio_speed = 10;

#else
//GPIO for push button
static int gpio_reset = 0;
//GPIO for virtual device
static int gpio_power = 16;
static int gpio_swing = 17;
static int gpio_mode_auto = 18;
static int gpio_mode_cool = 19;
static int gpio_mode_heat = 21;
static int gpio_speed = 22;
#endif

#define SECONDARY_I2C_PORT &Wire1

bool switch_state = true;

const char *service_name = "Thing01";
[... 154 lines cut ...]
          Serial.printf("Reset Wi-Fi.\n");
          wifi_connected = 0;
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        }
    }

    if (wifi_connected) {
      client.loop(); 

      temperature.updateAndReportParam("Temperature", rTemp);
      humidity.updateAndReportParam("Temperature", rHum);

      // Reconnect if disconnected
      if (client.connected()) {
        stream2API(rTemp, rHum);
      } else {
        connect();
      }
    }

    delay(10000);
}

```

shahpiyushv, 2022-10-03:
@chaudhariatul , do the default examples work well for you? If yes, one option is to wait for `RMAKER_MQTT_EVENT_CONNECTED` under the `RMAKER_COMMON_EVENT` and only then start your connection attempts. You can see [here](https://github.com/espressif/arduino-esp32/blob/a5f03a86512b270be1080d7e7cb82e4fd71c9a1a/libraries/RainMaker/src/RMaker.cpp#L41) for a sample of how to register event handlers.

chaudhariatul, 2022-10-03:
@shahpiyushv  Kindly share documentation link of all possible RMAKER events. If there are examples it will be very helpful. I wasn't able to find much here : https://rainmaker.espressif.com/docs/get-started.html

shahpiyushv, 2022-10-03:
@chaudhariatul , various RainMaker components use the ESP IDF's event framework and so the events are spread out into respective header files, the list for which can be found in the API docs.

[RMAKER_EVENT](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html#_CPPv418esp_rmaker_event_t)

[RMAKER_COMMON_EVENT](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_common.html#_CPPv425esp_rmaker_common_event_t)

[RMAKER_OTA_EVENT](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_ota.html#_CPPv422esp_rmaker_ota_event_t)

chaudhariatul, 2022-10-10:
@shahpiyushv are there any arduino examples?

chaudhariatul, 2022-10-10:
My guess is there are 2 issues:

1. Using  in the connect function is causing SSL errors to rainmaker cloud connection
```
  net.setCACert(CERT_CA);
  net.setCertificate(CERT_CRT);
  net.setPrivateKey(CERT_PRIVATE);
```

2. Rainmaker arduino library doesn't have RMAKER_COMMON_EVENT? I tried this example with arduino but made no progress. esp_event_handler_register(RMAKER_COMMON_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL)

shahpiyushv, 2022-10-10:
@chaudhariatul , I think the reason for the issue you are seeing is because of insufficient memory and hence, the suggestion to let provisioning finish and mqtt connect.

With regards to the event handler, is your event_handler function not getting called at all? Can you check if `esp_event_handler_register` is returning ESP_OK? Can you share your event handler function?