espressif/esp-rainmaker #181: add update OTA and Local control in Arduino IDE, please

Hassan-Z94, 2022-10-05:
i can't update with RMaker.enableOTA(OTA_USING_PARAMS);
How to update from the dashboard?
add local control In arduino IDE, please
.
.
There is no description of scene activation in Arduino... I was able to activate the scene with this method
.
must first -> #include <esp_rmaker_scenes.h>
Then in the setup ->esp_rmaker_scenes_enable(); -> before ->   RMaker.start();
With this method, the scene is activated in Arduino.
Thanks to all developers

sanketwadekar, 2022-10-05:
Hi @Hassan-Z94
Currently, the OTA feature has some issues, which will get fixed in the next release of Esp32 Arduino. Local control is not supported as of now. We will update you one this feature is released.
Thanks

sanketwadekar, 2022-12-23:
Hello,
Arduino 2.0.6 has been released, which fixes the OTA crash issues seen previously. If you face low memory errors like
esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00, please reboot your device once. Doing a reboot after successful provisioning and then starting OTA is recommended.
Thanks!