espressif/esp-rainmaker #182: Device becomes offline after router reboot

maxakash, 2022-10-09:
I have a reboot schedule for my wifi router everyday. After the reboot, the esp32 becomes offline. I have to power it off and on to make it online.

Any suggestions to resolve this problem.

@shahpiyushv 

Thanks

shahpiyushv, 2022-10-10:
@maxakash , is this with esp-idf or Arduino?

maxakash, 2022-10-10:
This is with Arduino.

sanketwadekar, 2022-10-10:
Can you try using the fix given [here](https://www.esp32.com/viewtopic.php?f=41&t=29011&p=101047&hilit=wifi&sid=ac5e266c42982ec8c02c9406771302a6#p101047)?

maxakash, 2022-10-10:
> Can you try using the fix given [here](https://www.esp32.com/viewtopic.php?f=41&t=29011&p=101047&hilit=wifi&sid=ac5e266c42982ec8c02c9406771302a6#p101047)?

Yes I have implemented it in my code. Will test and let you know. Closing this for now.

closed 2022-10-10 by maxakash