espressif/esp-rainmaker #183: IDF v5 compatibility

martinberlin, 2022-10-09:
I'm currently using since 2 months [IDF v5.0-beta1](https://github.com/espressif/esp-idf/releases/tag/v5.0-beta1) so far I could adapt my components and Firmware's without much problems to work in V5.
Tried so far with an old Lolin ESP32 and will try in the next days with an Espressif ESP32S3 dev board.
However I'm stuck with Rainmaker and so far could not make it work. 

First of all if I clone this repository, then the first error that comes is:

```
/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_utils.c:70:10: fatal error: esp_app_desc.h: No such file or directory
   70 | #include <esp_app_desc.h>
```

CMake is right in my esp-idf components directory, esp_app_format does not exist, hence also not  esp_app_desc.
To overcome this I added it as a component and further debug to the next problem:

It opens BLE, QR is not coming by serial, but using the URL: 
https://rainmaker.espressif.com/qrcode.html?data={%22ver%22:%22v1%22,%22name%22:%22PROV_d4ed0f%22,%22pop%22:%22eee0a04f%22,%22transport%22:%22ble%22}

I can see the QR and scan it with the ESP-Rainmaker app in Android. Then I get one of this errors:

1.

```
 I (1829) app_wifi: Provisioning will auto stop after 30 minute(s).
E (32299) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
```

or 

2. 

```
app_wifi: Provisioning Started. Name : PROV_d4ed0f, POP : eee0a04f
I (1752) app_wifi: Provisioning will auto stop after 30 minute(s).
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```

Can someone please enlighten me and explain what is the way to make Rainmaker work in ESP-IDF version 5?

shahpiyushv, 2022-10-10:
@martinberlin the best way is to pull the latest esp-rainmaker and all it's submodules (git submodule update --init --recursive), then pull the latest code on release/v5.0 branch of esp-idf, including all its submodules. Even then, you may see some failures/crashes during provisioning since [this commit](https://github.com/espressif/esp-idf/commit/1d8e7fe2d7549714d9f7a5cba288b5275d8a0ce7) in esp-idf master branch hasn't yet been backported on release/v5.0. You can manyally cherry-pick for the time being if required.

martinberlin, 2022-10-10:
Thanks @shahpiyushv that helped a bit. Now I can see the QR code. 

Updated idf branch to **release/v5.0** and all submodules like you described. 
I made that updates per hand since a git cherry pick didn't worked from release/v5.0 (object not good)

But still fails exaclty in the moment it connects with the app and tries to list the WiFi SSID's

```I (1583) app_wifi: Provisioning Started. Name : PROV_d4ed0f, POP : eee0a04f
I (1583) app_wifi: Provisioning will auto stop after 50 minute(s).
E (15863) security0: Security version mismatch. Closing connection
E (15863) protocomm_ble: Invalid content received, killing conn```

PROTOCOM Looks like this in menuconfig:
![Screenshot from 2022-10-10 14-45-07](https://user-images.githubusercontent.com/2692928/194869373-68e7e5db-4994-48c7-ac8c-b136e16db53c.png)

martinberlin, 2022-10-16:
Kept on trying but I must confess still no luck with IDF v5 and ESP Rainmaker.
If there is any news in the Espressif side please let me know and I will try again or I will try to track it down when I find some time.
But there is definitively something more than what @shahpiyushv pointed out.

martinberlin, 2022-10-17:
On a different topic that might be not really connected to this Issue but is still interesting for me in the IDF versionn 5 context is the following:
As an IDF developer what is the easiest way to add Rainmaker to an existing Firmware?

At the moment I'm adding it to my repository Cale that is an easy Firmware to download a Bitmap image in epaper displays. What I do at component level you can see here:

https://github.com/martinberlin/cale-idf/tree/feature/50-idf-v5-rainmaker/components
Basically I copy **esp_rainmaker** latest version from here. And then all what can be linked as github submodules like the ones that are being used, instead of copying them I also add them to mine (.gitmodules)

```
[submodule "components/rmaker_common"]
	path = components/rmaker_common
	url = https://github.com/espressif/esp-rainmaker-common.git
[submodule "components/json_parser/upstream"]
	path = components/json_parser/upstream
	url = https://github.com/espressif/json_parser.git
[submodule "components/json_generator/upstream"]
	path = components/json_generator/upstream
	url = https://github.com/espressif/json_generator.git
```

This works but is a bit cumbersome to maintain and probably there is a better way to do it. 
As a developer my wish would be to add Rainmaker as a single repository (With it's submodule dependencies, for sure) and then that I can use to make my Rainmaker-ize my program and use it as the only way to make WiFi provisioning.
For me would be ideal, that it also provides any kind of smart function, where you can pass your **drawFunction(x, y, color)** and it draws the QR code for you does not matter what library you use.  
I think most of the people doing Firmware's that use displays will greatly appreciate something like this. So you can give out the finished product with a QR code printed in the screen instead of the HTML QR code in a printed page.

shahpiyushv, 2023-01-13:
@martinberlin Sorry for missing this out, but let me provide some answers here

1. The best way to integrate RainMaker components is to add the esp-rainmaker repository itself as a submodule, set RMAKER_PATH to point to the esp-rainmaker like you can see in the [examples](https://github.com/espressif/esp-rainmaker/blob/master/examples/fan/CMakeLists.txt#L5) and then modify the CMakeFile to include all relevant paths of rainmaker components like you can see [here](https://github.com/espressif/esp-rainmaker/blob/fe94cc6994c4d012bad3ce04d6ae07b616894714/examples/fan/CMakeLists.txt#L12)

2. If you disabled user node association, the device cannot be provisioned using ESP RainMaker app, since user association is a mandatory step for RainMaker. However, you can use the [ESP Provisioning apps](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html#provisioning-tools). The device will no more have an end user, but can still have an admin user, who can access the node from the dashboard.
3. You can disable the on screen QR code display (using CONFIG_APP_WIFI_PROV_SHOW_QR ) and then use APP_WIFI_EVENT_QR_DISPLAY to display it on your own.

martinberlin, 2023-01-13:
Great some I'm closing here and marking it "solved" in the title. Thanks @shahpiyushv

closed 2023-01-13 by martinberlin